ifm/ifm3d-ros #2: ifm3-ros crashes when running roslaunch ifm3d camera.launch

graugans, 2017-09-07:
On a freshly set-up Ubunbtu 16.04 laptop with the latest ifm3d lib (0.3.1) a O3X100 running a 0.1.20 software version.

I get this
```
[ INFO] [1504771124.041250388]: Running dtors...
[ INFO] [1504771124.041318544]: Initializing camera...
[ INFO] [1504771125.056067573]: Initializing framegrabber...
[ INFO] [1504771126.493181174]: Initializing image buffer...
[ WARN] [1504771126.994236347]: Timeout waiting for camera!
[ INFO] [1504771127.082569353]: Got unit vectors, restarting framegrabber with mask: 15
[ INFO] [1504771127.082744506]: Running dtors...
*** Error in `/home/rosuser/ros/ifm3d/lib/ifm3d/ifm3d_node': corrupted double-linked list (not small): 0x0000000001febf00 ***
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x777e5)[0x7f06bc2617e5]
/lib/x86_64-linux-gnu/libc.so.6(+0x7d814)[0x7f06bc267814]
/lib/x86_64-linux-gnu/libc.so.6(+0x81008)[0x7f06bc26b008]
/lib/x86_64-linux-gnu/libc.so.6(cfree+0x4c)[0x7f06bc26e53c]
/usr/lib/libifm3d_framegrabber.so.0.3.1(_ZN5ifm3d12FrameGrabber4ImplD1Ev+0x148)[0x7f06bcd708b8]
/usr/lib/libifm3d_framegrabber.so.0.3.1(_ZN5ifm3d12FrameGrabberD1Ev+0x33)[0x7f06bcd70cb3]
/home/rosuser/ros/ifm3d/lib/ifm3d/ifm3d_node(_ZN9__gnu_cxx13new_allocatorIN5ifm3d12FrameGrabberEE7destroyIS2_EEvPT_+0x23)[0x47eaa3]
/home/rosuser/ros/ifm3d/lib/ifm3d/ifm3d_node(_ZNSt16allocator_traitsISaIN5ifm3d12FrameGrabberEEE7destroyIS1_EEvRS2_PT_+0x23)[0x47d587]
/home/rosuser/ros/ifm3d/lib/ifm3d/ifm3d_node(_ZNSt23_Sp_counted_ptr_inplaceIN5ifm3d12FrameGrabberESaIS1_ELN9__gnu_cxx12_Lock_policyE2EE10_M_disposeEv+0x37)[0x47bd17]
/home/rosuser/ros/ifm3d/lib/ifm3d/ifm3d_node(_ZNSt16_Sp_counted_baseILN9__gnu_cxx12_Lock_policyE2EE10_M_releaseEv+0x42)[0x469a26]
/home/rosuser/ros/ifm3d/lib/ifm3d/ifm3d_node(_ZNSt14__shared_countILN9__gnu_cxx12_Lock_policyE2EED2Ev+0x27)[0x4642f3]
/home/rosuser/ros/ifm3d/lib/ifm3d/ifm3d_node(_ZNSt12__shared_ptrIN5ifm3d12FrameGrabberELN9__gnu_cxx12_Lock_policyE2EED2Ev+0x1c)[0x45ef36]
/home/rosuser/ros/ifm3d/lib/ifm3d/ifm3d_node(_ZNSt12__shared_ptrIN5ifm3d12FrameGrabberELN9__gnu_cxx12_Lock_policyE2EE5resetEv+0x46)[0x466060]
/home/rosuser/ros/ifm3d/lib/ifm3d/ifm3d_node(_ZZN9IFM3DNode3RunEvENKUltE_clEt+0x192)[0x460b40]
/home/rosuser/ros/ifm3d/lib/ifm3d/ifm3d_node(_ZN9IFM3DNode3RunEv+0xa61)[0x4621ef]
/home/rosuser/ros/ifm3d/lib/ifm3d/ifm3d_node(main+0xb0)[0x45b5ac]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf0)[0x7f06bc20a830]
/home/rosuser/ros/ifm3d/lib/ifm3d/ifm3d_node(_start+0x29)[0x45b2b9]
======= Memory map: ========
00400000-0049b000 r-xp 00000000 08:01 1577746                            /home/rosuser/ros/ifm3d/lib/ifm3d/ifm3d_node
0069a000-0069b000 r--p 0009a000 08:01 1577746                            /home/rosuser/ros/ifm3d/lib/ifm3d/ifm3d_node
0069b000-0069c000 rw-p 0009b000 08:01 1577746                            /home/rosuser/ros/ifm3d/lib/ifm3d/ifm3d_node
0069c000-0069d000 rw-p 00000000 00:00 0 
01ca0000-021ee000 rw-p 00000000 00:00 0                                  [heap]
7f068c000000-7f068c022000 rw-p 00000000 00:00 0 
7f068c022000-7f0690000000 ---p 00000000 00:00 0 
7f0690000000-7f0690085000 rw-p 00000000 00:00 0 
7f0690085000-7f0694000000 ---p 00000000 00:00 0 
```

tpanzarella, 2017-09-07:
Hi Christian, Thanks for reporting this. I will look into it (FYI, I'm at a customer all day, so, will need to investigate tonight)

Quick question, is this consistent and repeatable?

graugans, 2017-09-07:
@tpanzarella it is reproducible by repeating the call ``roslaunch ifm3d camera.launch`` I've not tried with another camera or PC. Take your time.

tpanzarella, 2017-09-08:
@graugans I have been spending some time on this today and I cannot seem to duplicate it. Beyond trying on my own machine, I have brought up a fresh version of Ubuntu 16.04 and a fresh version of 14.04. Neither allowed me to duplicate this issue. For clarity, I followed exactly [these steps for 16.04](https://github.com/lovepark/ifm3d-ros/blob/master/doc/kinetic.md) and exactly [these steps for 14.04](https://github.com/lovepark/ifm3d-ros/blob/master/doc/indigo.md). Beyond the "straight forward" testing, I also tried running multiple instances of the ROS node against a single camera hoping that the resource contention could perhaps cause me a problem. However, the behavior was as I expected and did not cause any issues (i.e., the ROS node is designed to expect such hostility).

All of the above being said, we need to figure this out and I have a couple of ideas:

**1. Mismatched versions of OpenCV**
We need to ensure that both the ROS node and ifm3d are linking to the same version of OpenCV. Can you check that? Probably the quickest way would be to use `ldd`. My expectations would be to see something like this on 16.04 with Kinetic:
```
$ roscd ifm3d/
$ cd ../../lib/ifm3d/
$ ldd ifm3d_node | grep -i 'opencv'
	libopencv_core3.so.3.2 => /opt/ros/kinetic/lib/libopencv_core3.so.3.2 (0x00007fc11a240000)
	libopencv_imgcodecs3.so.3.2 => /opt/ros/kinetic/lib/libopencv_imgcodecs3.so.3.2 (0x00007fc1170bc000)
	libopencv_imgproc3.so.3.2 => /opt/ros/kinetic/lib/libopencv_imgproc3.so.3.2 (0x00007fc115905000)

$ ldd /usr/lib/libifm3d_image.so | grep -i 'opencv'
	libopencv_core3.so.3.2 => /opt/ros/kinetic/lib/libopencv_core3.so.3.2 (0x00007fd97a00e000)
```
Looking at this could rule out the mismatched OpenCV version issue.

**2. Rebuilding the ROS node**
I'm sure you did this already but, after you updated `ifm3d` to `0.3.1`, did you rebuild and relink the ROS node to the new version? This would be another thing to try if this is not the case. Although, you mentioned a "fresh install", so this may be a moot point.

**3. The Timeout log message**
If the above two issues check out, the only other thing I can think of is to follow the clue from your logs, specifically, this snippet:
```
[ INFO] [1504771126.493181174]: Initializing image buffer...
[ WARN] [1504771126.994236347]: Timeout waiting for camera!
[ INFO] [1504771127.082569353]: Got unit vectors, restarting framegrabber with mask: 15
[ INFO] [1504771127.082744506]: Running dtors...
```
The `Timeout...` message being listed there is not normal. It also concerns me on the O3X as we have to deal with the Unit Vectors differently than we do on the O3D and there is a little bit of "memory shuffling" that we do to make the O3X look like the O3D inside of the framegrabber at this point. So, a few additional questions:

1. Prior to the crash, do you consistently see that `Timeout...` log message?
2. If you take ROS out of the loop, can you use the `$ ifm3d viewer` application and see the O3X point cloud OK?

OK, let's see if looking at this stuff uncovers any other clues to get this sorted. Thanks in advance for the help in troubleshooting -- without being able to duplicate this issue I feel a bit stuck in terms of code to look at / fix.

graugans, 2017-09-09:
```
@graugans I have been spending some time on this today and I cannot seem to duplicate it. Beyond trying on my own machine, I have brought up a fresh version of Ubuntu 16.04 and a fresh version of 14.04. Neither allowed me to duplicate this issue. For clarity, I followed exactly these steps for 16.04 and exactly these steps for 14.04. Beyond the "straight forward" testing, I also tried running multiple instances of the ROS node against a single camera hoping that the resource contention could perhaps cause me a problem. However, the behavior was as I expected and did not cause any issues (i.e., the ROS node is designed to expect such hostility).
```
I`ve installed ROS as they have described it on their website. It was a few weeks ago. 

```
All of the above being said, we need to figure this out and I have a couple of ideas:

1. Mismatched versions of OpenCV
We need to ensure that both the ROS node and ifm3d are linking to the same version of OpenCV. Can you check that? Probably the quickest way would be to use ldd. My expectations would be to see something like this on 16.04 with Kinetic:

$ roscd ifm3d/
$ cd ../../lib/ifm3d/
$ ldd ifm3d_node | grep -i 'opencv'
	libopencv_core3.so.3.2 => /opt/ros/kinetic/lib/libopencv_core3.so.3.2 (0x00007fc11a240000)
	libopencv_imgcodecs3.so.3.2 => /opt/ros/kinetic/lib/libopencv_imgcodecs3.so.3.2 (0x00007fc1170bc000)
	libopencv_imgproc3.so.3.2 => /opt/ros/kinetic/lib/libopencv_imgproc3.so.3.2 (0x00007fc115905000)

$ ldd /usr/lib/libifm3d_image.so | grep -i 'opencv'
	libopencv_core3.so.3.2 => /opt/ros/kinetic/lib/libopencv_core3.so.3.2 (0x00007fd97a00e000)

Looking at this could rule out the mismatched OpenCV version issue.
```
I'll check this when I can get my hand on the laptop again.

```
2. Rebuilding the ROS node
I'm sure you did this already but, after you updated ifm3d to 0.3.1, did you rebuild and relink the ROS node to the new version? This would be another thing to try if this is not the case. Although, you mentioned a "fresh install", so this may be a moot point.
```
So there was no ROS node with the older version of the lib. 

```
3. The Timeout log message
If the above two issues check out, the only other thing I can think of is to follow the clue from your logs, specifically, this snippet:

[ INFO] [1504771126.493181174]: Initializing image buffer...
[ WARN] [1504771126.994236347]: Timeout waiting for camera!
[ INFO] [1504771127.082569353]: Got unit vectors, restarting framegrabber with mask: 15
[ INFO] [1504771127.082744506]: Running dtors...

The Timeout... message being listed there is not normal. It also concerns me on the O3X as we have to deal with the Unit Vectors differently than we do on the O3D and there is a little bit of "memory shuffling" that we do to make the O3X look like the O3D inside of the framegrabber at this point. So, a few additional questions:

    Prior to the crash, do you consistently see that Timeout... log message?
```
This timeout thing made me stuck as-well. I will test this with another installation soon.
```
    If you take ROS out of the loop, can you use the $ ifm3d viewer application and see the O3X point cloud OK?
```
Yes the viewer was working really nice
```
OK, let's see if looking at this stuff uncovers any other clues to get this sorted. Thanks in advance for the help in troubleshooting -- without being able to duplicate this issue I feel a bit stuck in terms of code to look at / fix.
```

graugans, 2017-09-19:
The issue was caused by installing *ifm3d* before installing ros. This caused to link against the wrong version of OpenCV. Thanks for you help @tpanzarella

closed 2017-09-19 by graugans