ifm/ifm3d-ros #3: Nodelet interface

madsherlock, 2017-11-22:
Hello,

Is a Nodelet interface under development?
It would be a benefit on any system processing 2D and 3D data, especially with high resolution settings.

My idea is it could simply hold a std::shared_ptr<IFM3DNode> allocated upon Nodelet initialization.
I am unsure about whether this would work with the ros::AsyncSpinner.
Any ideas?

Best regards
Mikael

tpanzarella, 2017-11-22:
We do have plans to re-implement as a nodelet -- similar to our [Royale Interface](https://github.com/lovepark/royale-ros), however, to answer your questions directly, that work is *not* _under development_ as of yet.

madsherlock, 2017-11-22:
Thanks!
I will let you know if I succeed in making the proposed Nodelet (Node wrapper).

closed 2017-11-22 by madsherlock