ifm/ifm3d-ros #4: Exposure time of individual pixels

madsherlock, 2017-11-28:
Hello,

I  was under the impression that exposure times could be registered to individual pixels by inspecting bits 5 and 4 of the confidence image. However, running an O3D in dual exposure mode, both bits are unset, indicating "unused".
Even if I had placed the camera in single exposure mode, I would expect both bits to be set.

Is there a way to enable the setting of bits 4 and 5 of the confidence image?
I tried setting the "schema_mask" parameter to enable the transfer of the exposure time list using 'EXP_TIME'. Did not fix problem.

Basically, I would like to have enough information to normalize the amplitude image myself. Is this possible?

Best regards
Mikael

graugans, 2018-01-10:
Here is the definition of the confidence bits:

|Bit| Meaning when bit is set|
|---|-------------------------------|
|0 | **Pixel is invalid** To determine if a pixel is valid or not, only this bit needs to be checked. The distance values, as well as the Cartesian coordinates of invalid pixels are set to 0. The reason for invalidity is encoded in some of the other bits below. |
|1| **Saturation** Signal strength of the ambient and/or modulated light exceeds the upper limit.|
|2| **Implausible measurement** Four phase validation of this pixel failed due to movement during the exposure or other kind of disturbances like an interfering modulated light source in the observed scene.|
|3| **Low amplitude** Integrated signal strength of the own modulated light was not sufficient for a  reliable distance measurement.|
|4| **Long exposure time taken** In double exposuremodesthisbitindicateswhetherthe long or the short integration time was selected.|
|5| **Double frequencymeasurementmismatch** In double frequency modes this bit indicates a implausible measurement due to a mismatch of the two frequency distance values. Fast movement in the scenery can be a reason for this bit to be set.|
|6| **Reserved** |
|7| **Reserved** |

madsherlock, 2018-01-14:
Thanks! Where did you get this information?
I have the Operating Instructions for the O3D3xx, and they list a different table from yours.

Bit 5, bit 4:
0 0 = unused
0 1 = shortest exposure time (only used in 3 exposure mode)
1 0 = middle exposure time in 3 exposure mode, short exposure in double exposure mode
1 1 = longest exposure time (always 1 in single exposure mode)

Bit 6: 1 = pixel is clipped

Bit 7: Suspect/defective pixel   (interpolated from surrounding pixels)

From this, I would expect the combination I'm seeing, 0 0, never to appear.
However, with your table, this combination would appear whenever the long exposure time was used, or when I'm in single exposure mode.

theseankelly, 2020-01-13:
Closing due to age of issue -- please reactivate or open a new issue if further assistance is required!

closed 2020-01-13 by theseankelly