ifm/ifm3d-ros #9: Package rename

willcbaker, 2018-09-27:
I would like to propose a package rename. 

The issue I have is that in the Cmake file we define the package name and call find_package() on the exact package name: https://github.com/lovepark/ifm3d-ros/blob/master/CMakeLists.txt#L4

According to ROS REP-144 http://www.ros.org/reps/rep-0144.html
> a package name should not contain ros as it is redundant. Exceptions include core packages and ROS bindings of an upstream library (e.g. moveit_ros)

I think this is a valid exception; as we compile against an upstream library (ifm3d @ https://github.com/lovepark/ifm3d)

I have already renamed the package in a private repo as it was the only way to accommodate a CI system (package name conflicts), and can submit a PR upon request.

tpanzarella, 2018-09-27:
IIRC, @graugans has something working as well as he is working on getting both this project and [ifm3d](https://github.com/lovepark/ifm3d) into the core ROS distribution. Maybe he can shed some light on what he did -- I think it was actually renaming the underlying driver to something like *ifm3d_core* or something like that. 

From a user interface perspective, I like being able to say:

```
$ roslaunch ifm3d camera.launch
```

As opposed to something lke:

```
$ roslaunch ifm3d_driver camera.launch
```

Hopefully @graugans has some perspective here. Also, send in your PR so we can take a look at what you propose. It will give us a basis for a concrete discussion around this topic. You make a good point, let's land upon a solution we are all happy with. Realize, there are some companies out there embedding this stuff into their products and they likely have scripts and other tooling in place that assumes the name `ifm3d`. We just need to contemplate this in our discussion as well.

graugans, 2018-09-28:
@willcbaker and @tpanzarella I agree on the fact that this is kind of an issue that both packages are named ifm3d. As discussed in this issue on [ifm3d](https://github.com/lovepark/ifm3d/issues/93#issuecomment-420571593) from ifm3d the idea is to have at-least the ifm3d library package called ifm3d-core. For the ROS driver I do not see any need to rename it. Because it is already called ``ifm3d`` within ROS. The postfix ``-ros`` is only used in the repository name.

According to what @eggerk wrote in https://github.com/lovepark/ifm3d/issues/93 issue [#93](https://github.com/lovepark/ifm3d/issues/93) I do have to change the installation path of the package.xml in my [patch](https://github.com/graugans/ifm3d-release/blob/patches/release/kinetic/ifm3d-core/0001-Added-install-rule-for-catkkin-package.xml.patch) so there is no conflict with the ``ifm3d-ros`` driver installation. I will try to polish up my https://github.com/graugans/ifm3d-release repo so it is ready to be added to the ros-distro.

But due to the fact I am quite new to all this bloom and ROS release process I am always interested in advice and suggestions. So @willcbaker please share your ideas with us. I tried to keep the original names as much as possible because renaming stuff may will break use-cases of non ROS users.

willcbaker, 2018-10-22:
@graugans how is the release for ifm3d-core going for you?
If I understand correctly, you plan to release ifm3d on the ros distro under the name ifm3d-core and thus resolve the name conflict without changing either package?

graugans, 2018-10-24:
@willcbaker The release process is making progress. Today I pushed couple of bug-fixes to the [ifm3d-release](https://github.com/ifm/ifm3d-release) repo. So what works for me is this:

#Building a ifm3d-core package

```
$ git clone https://github.com/graugans/ifm3d-release.git
$ cd ifm3d-release
$ git checkout debian/kinetic/xenial/ifm3d-core
$ bloom-generate rosdebian --ros-distro kinetic
$ fakeroot debian/rules binary
```
Installing the generated package
```
# Assuming we are still in the ifm3d-release folder
$ sudo dpkg -i ../ros-kinetic-ifm3d-core_0.11.0-0xenial_amd64.deb
```

After this the ifm3d-ros package can be build
```
$ git clone https://github.com/lovepark/ifm3d-ros.git
$ cd ifm3d-ros
$ bloom-generate rosdebian
$ fakeroot debian/rules binary
$ sudo dpkg -i ../ros-kinetic-ifm3d_0.6.0-0xenial_amd64.deb
```

Just as a information the content of the you might need to source the ``setup.bash`` before executing the commands above
```
$ source /opt/ros/kinetic/setup.bash
```

theseankelly, 2020-01-13:
This work is now being tracked by #30 -- 'submit ifm3d to the ROS package index'

closed 2020-01-13 by theseankelly