ifm/ifm3d-ros #17: Free space pointcloud / points at infinity

fabianbl, 2019-02-26:
I would like to get a free space pointcloud from the 03x100, assuming that all invalid pixels are free space. We need this to clear free space in an occupancy grid. I saw that right now, all invalid pixels are at the origin in the pointcloud, and there is no way to recover the direction of the ray. Is there an option to change this?

tpanzarella, 2019-02-27:
I think you have several options. First, the unit vectors are available on a latched topic. Second, you are correct, pixels flagged as invalid in the confidence image are explicitly set to 0 -- which is "impossible" so those values can be used as sentinels to replace with NaN, Infinity, or whatever you want -- or you could also consult the confidence image yourself to set those pixels to whatever you would like. If you would like to handle this at the driver level, you can write your own image container that handle invalid pixels however you would like. Details on writing an image container is [here](https://github.com/ifm/ifm3d/blob/master/doc/img_container.md). However, you would also need to write a ROS projection of that as well, which is likely more work than you want to take on.

Hopefully the above suggestions work for you.

graugans, 2019-03-05:
@fabianbl can you please close the issue when your problem is solved?

closed 2019-03-05 by fabianbl