ifm/ifm3d-ros #18: rosdep integration

wixiw, 2019-04-09:
J'ai tenté de compiler le driver ROS d'ifm pour le 03D dans mon workspace ROS et j'ai une erreur : 
/usr/include/ifm3d/fg/detail/byte_buffer.hpp:21:26: fatal error: glog/logging.h: No such file or directory

La correction semble évidente, il me manque une dépendance de l'os "glog" que l'auteur de : 
https://github.com/ifm/ifm3d-ros/blob/master/doc/building.md 
doit supposé être présente après des commandes proposées par :
https://github.com/ifm/ifm3d-ros/blob/master/doc/kinetic.md
Notamment : sudo apt-get install libgoogle-glog-dev
Mais  dans l'univers ROS on s'attend à ce que l'outil rosdep fasse le travail, car la commande suivante me dit que tout va bien : 
rosdep install --from-paths src --ignore-src -r -y
#All required rosdeps installed successfully

A priori il semble que les dépendances que vous souhaitez ne soient pas déjà incluses, 
 rosdep resolve google-glog
ERROR: no rosdep rule for 'google-glog'
rosdep resolve xmlrpc-c++8
ERROR: no rosdep rule for 'xmlrpc-c++8'

il faut donc demander à ce qu'elles le soient : 
http://docs.ros.org/independent/api/rosdep/html/contributing_rules.html  

L'environement ROS a pour objectif de permettre de facilement partager des codes sources sans qu'il soit nécessaire de se plonger dans la plomberie de chaque paquet. Il est donc important que l'on puisse installer le paquet ifm3d-ros avec les seuls outils de ROS.

graugans, 2019-04-09:
Hey @wixiw could you please translate the above to English please. Maybe @Arag24 can assist you

Arag24, 2019-04-09:
Hi @wixiw , @graugans ,
Question below is related to inclusion of libraries source when building `ifm3d-ros`.

@wixiw could see from [https://github.com/ifm/ifm3d-ros/blob/master/doc/kinetic.md](https://github.com/ifm/ifm3d-ros/blob/master/doc/kinetic.md)
it's mandatory to include :

> $ sudo apt-get install libxmlrpc-c++8-dev
$ sudo apt-get install libgoogle-glog-dev

However, as he was using `rosdep` command, and the answer was

>  #All required rosdeps installed successfully

I'm pretty sure he spent some time to find what dependency he has to includewith, is that true @wixiw?

Open topic is the following : 
As there is no `ERROR: no rosdep rule for 'google-glog'`, @wixiw is asking if rules can be added directly from here [(http://docs.ros.org/independent/api/rosdep/html/contributing_rules.html)](http://docs.ros.org/independent/api/rosdep/html/contributing_rules.html).
In such way, `ifm3d-ros` would be more efficient, having dependecy included without having deep look inside ifm repo, and only check with command as 

> rosdep install --from-paths src --ignore-src -r -y
to see if eveything is running correctly ? (as apparently many should do in ROS environment)

Does it make sense, @graugans , @wixiw  ?

wixiw, 2019-04-11:
I translated the question. @Arag24 perfect translation of the need.

Note that I (obviously) have a workaround from the "apt-get" commands, I just send the request for your continuous improvment.

theseankelly, 2020-01-13:
This work is now being tracked by #30 -- 'submit ifm3d to the ROS package index'

closed 2020-01-13 by theseankelly