ifm/ifm3d-ros #19: Unable to change configurations from dump file

derektan95, 2019-06-12:
Hi, 

I noticed some unexpected noise from my image and I figured from the ifm VisionAssistant software that this noise could be removed if I were to change the "Maximum Visible Distance" settings from "Less than 5m[1 frequency] to Up to 30m[2 frequencies].
[image]

I was following this manual here in this link: https://github.com/ifm/ifm3d-ros/blob/master/doc/dump_and_config.md

Unfortunately, I got this error which says that there isn't a "dump" executable. I wonder why I am having this error? Thanks!

[image]

tpanzarella, 2019-06-12:
It looks like you did not `catkin_make install` your workspace (and also source in the installed `setup.bash` file). The `dump` script pointed to by the path shown in your shell is from the source distro where the executable bit on the file is *not* set. To use the ROS interface as it was intended, you need to *install* the package -- these scripts, with their executable bits set, will end up installed as `${CMAKE_INSTALL_PREFIX}/lib/ifm3d/dump` and `${CMAKE_INSTALL_PREFIX}/lib/ifm3d/config`. This will allow `rosrun ifm3d ...` to properly find them as intended. Also realize, these ROS versions of `dump` and `config` necessarily require the camera nodelet to be running as they simply proxy request/responses between your shell and the exposed `~/Dump` and `~/Config` ROS services exposed by the node.

Alternatively, to achieve what you want, you can also just use the standard, ROS-independent [ifm3d command line program](https://github.com/ifm/ifm3d/blob/master/doc/configuring.md), which, should be installed on your `$PATH` at `/usr/bin/ifm3d` if you installed from `deb` or built (and installed) according to the recommended instructions [here](https://github.com/ifm/ifm3d/blob/master/doc/source_build.md). For clarity, the `ifm3d` command line tool is part of the ifm3d `tools` module, which the ROS wrapper does *not* depend on. So, you will have had to installed (or built) the `tools` deb.

HTH.

derektan95, 2019-06-19:
Sounds good! Thank you so much for your detailed comments.

closed 2019-06-19 by derektan95