ifm/ifm3d-ros #25: How to export point cloud data into .pcd and .ply files ?

arunumd, 2019-10-01:
I have been trying to generate .ply files of the point cloud data for numpy processing using [pcl_ros package](http://wiki.ros.org/pcl_ros) and also the [pcl_pcd2ply](http://manpages.ubuntu.com/manpages/bionic/man1/pcl_pcd2ply.1.html)  command. The .ply files seem to open in Meshlab properly. But for some reason, the same .ply files wouldn't get loaded in numpy using [plyFile](https://github.com/dranjan/python-plyfile) and I keep getting [duplicate properties](https://stackoverflow.com/q/58183252/10019073) error. Now I am suspicious whether the way in which the .ply file was generated had any negative effect on the file structure ?

Would you be able to suggest an effective way to generate .ply files (ascii encoding) of the pallet detection point cloud outputs ?

theseankelly, 2019-10-02:
Hey Arun -- I see you're using the PDS camera. Are you obtaining your point clouds from the pds_client ROS node, or from the ifm3d-ros node? If you're working through PDS I suggest we continue forward on our email thread. Otherwise (for generic ifm3d-ros support) we can continue the dialogue here.

arunumd, 2019-10-02:
Hi @theseankelly -- Glad to assist you with more details. The [pcl_ros package](http://wiki.ros.org/pcl_ros) seems to be generating a substandard pcd file from the rosbag topic. This in turn has a domino effect when we export the same pcd file to different point cloud formats (ply, etc.). Shown below is the pcd file header of the one generated by the [pcl_ros package](http://wiki.ros.org/pcl_ros) :
```
# .PCD v0.7 - Point Cloud Data file format
VERSION 0.7
FIELDS x y z _ intensity _
SIZE 4 4 4 1 4 1
TYPE F F F U F U
COUNT 1 1 1 4 1 12
WIDTH 352
HEIGHT 264
VIEWPOINT 0 0 0 1 0 0 0
POINTS 92928
DATA binary
```

Like I mentioned in the email, there are two blank spaces in the fields area which are the cause of the issue. I am trying to figure out how to address these blank spaces which don't denote any specific name.

arunumd, 2019-10-02:
> Hey @arunumd -- I see you're using the PDS camera. Are you obtaining your point clouds from the pds_client ROS node, or from the ifm3d-ros node? If you're working through PDS I suggest we continue forward on our email thread. Otherwise (for generic ifm3d-ros support) we can continue the dialogue here.

I am working with the ifm3d-ros node from the rosbag files generated by the data collection utility in the PDS - VM.

arunumd, 2019-10-02:
The following are the steps we followed (commands on the terminal) to solve  the problem :
```
rosrun pcl_ros bag_to_pcd <input_file.bag> <topic> <output_directory>
cd output_directory
pcl_converter -f ascii input_fine_name.pcd output_file_name.pcd
pcl_pcd2ply -format 0 input_file_name.pcd output_file_name.ply
```

closed 2019-10-02 by arunumd

Cristian-wp, 2022-10-27:
Hi @arunumd thanks for your explanation.
I have follow your instructions but from the rosbag I get 633 singles files .pdc instead of complete file.
I am doing somethin wrong?

lola-masson, 2022-11-09:
What you are describing is the default behavior of the pcl converter, as far as I can tell. Would something like the cloud assembler mentioned [here](https://answers.ros.org/question/235063/bag-file-to-pcd-files/) solve your problem?

Cristian-wp, 2022-11-09:
I think it can solve the problem. I will try and let you know the results.

graugans, 2022-11-09:
> I think it can solve the problem. I will try and let you know the results.

Hej, @Cristian-wp could you please be so kind to share the solution with this community.

Cristian-wp, 2022-11-09:
At the moment I am working on another project. I think in these days I will try and I will share the solution.