ifm/ifm3d-ros #27: How to change the frame rate of rostopic ?

zhaoyingdeveloper, 2019-11-27:
Hi guys,
where can I set default frame rate of rostopic like "/ifm3d/camera/cloud" ? I want to speed up pointcloud output. Thanks!

graugans, 2019-11-27:
You can change the parameters of the device through the ``Config`` service: https://github.com/ifm/ifm3d-ros#advertised-services 

You can use the [ifm3d](https://github.com/ifm/ifm3d) ``hz`` commandline tool for measuring the current framerate:
```
ifm3d hz --nframes=100
```

zhaoyingdeveloper, 2019-11-28:
Hi @graugans ,

Sorry for bother you again, I have already changed the Config through your guide and set the frame rate to 30(img1), but I find that the real frame is still low(img2 & img3) around 10. 

I set the framerate to 30:
img1:
[image]

But the truth framerate is around 10:
img2:
[image]

img3:
[image]

I want to speed up the pointcloud output frame rate greater than 20, how to solve it ?

Best Regards,

Carolina

graugans, 2019-11-28:
Nor worries Carolina, the framerate you are setting is the maximum framerate. There are some limitations which can lead to lower framerates. In your dump from above there is a MaxAllowedLEDFrameRate which is significantly lower than the desired framerate of 30. This is because of eye-safety and to prevent the LEDs from getting too hot. Depending on the mode you've selected the framerate drops to lower values because there are more images to be taken and calculated internally. Depending on your scene you could try to:
- Lower your exposure time
- Change the mode to a ``_low`` mode

My main question is why do you need 30Hz?

zhaoyingdeveloper, 2019-11-28:
Thank you for your reply @graugans,  since the algorithm may take some time, to avoid the final output frame being too low,  I want the source input pointcloud frame to be high.

In addition,  "lower the exposure time" is really helpful,  but what do you mean change the mode to a _low mode, I didn't find where to set the mode?

graugans, 2019-11-29:
You can get a list of supported modes like this:
```
ifm3d@7bc03a89a811:~$ ifm3d imager-types
[
  "morethan30m_low",
  "morethan30m_moderate",
  "under5m_high",
  "under5m_low",
  "under5m_moderate",
  "upto30m_high",
  "upto30m_low",
  "upto30m_moderate"
]
```
To change the mode you can manipulate the JSON config:
```
ifm3d dump | jq '.ifm3d.Apps[0].Imager.Type="under5m_moderate"' | ifm3d config
```
And check if it was succesful:
```
ifm3d@7bc03a89a811:~$ ifm3d dump | jq '.ifm3d.Apps[0].Imager.Type'
"under5m_moderate"
```
ifm3d dump | jq '.ifm3d.Apps[0].Imager.FrameRate="10"' | ifm3d config

closed 2019-11-30 by zhaoyingdeveloper