ifm/ifm3d-ros #28: the meaning of parameters in "Camera and Imager Configuration" ?

Carolinadeveloper, 2020-01-02:
Hi guys,

Is there any docs that describe the meaning or function of parameters in "Camera and Imager Configuration" [()](https://github.com/ifm/ifm3d/blob/master/doc/configuring.md)?

My camera type is O3D303,

Since I want to speed up the frame rate of pointcloud, I changed the following three parameters ( kept other parameters as original setting):

1. "ExposureTime" to "500"
2. "FrameRate" to "30"
3. "Type" to  "under5m_low"

Now I speed up the frame rate as expected but there exists disturbance(slight data change) which directly affects the stability. 

For example, when I changed the "ExposureTime" to "500", "800" and "1000",  the normals of the wall in front of the camera(distance is 0.9 meters) gives different degrees and differences up to 2 degrees, but the wall is quite and still, the camera is also fixed.

My question:
1. whether the pointcloud instability is due to the modification of the three camera configuration that I talked above?
2. How to set the parameters to get a correct and stable point cloud and why the exposure time may influence the result?(as my example, a stable and correct normal is expected)
3. Is the ray of light may lead to the disturbance? (what parameters should I modify based on different light situation)

GreNait, 2020-01-08:
Hi @Carolinadeveloper,

All three parameters you mentioned have an influence on the maximal possible framerate and the quality of the pointcloud. In principle, a higher framerate leads to less stable pointclouds. 

To get the most stable pointcloud a higher exposure time is key. As higher the exposure time, as more light we can receive - therefore better and more stable measurements. However, a higher exposure time can also lead to problems with reflective objects. Like white walls. This is the reason, why we integrated a double (or triple) exposure time mode. So you can choose a higher and a lower exposure time - one for bright objects and one for dark ones. I recommend this feature, but I am not sure about your usecase.
The problem is, that a double exposure leads to half of the possible framerate. 30fps will be cut to 15fps.

"under5m_low" is a setting responsible for the ambiguous range issues with ToF cameras. If you max. distance in a image is less than 5m, your good to go. If not, consider using a higher mode. A higher mode leads also to lower framerate. The next higher mode (up to 30m) cuts the framerate in half again. From the 15fps (considering double exposure) to around 7fps.

The ifm provides a manual for the ifmVisionAssistant, where the modes and settings are described. Perhaps this also helps to understand the relation between the settings. I include the link to it. From page 42 onward are some helpful descriptions.

Good luck with it :) and keep rockin.

[](url)https://www.ifm.com/mounting/706395UK.pdf

Carolinadeveloper, 2020-01-15:
> Hi @Carolinadeveloper,
>
> All three parameters you mentioned have an influence on the maximal possible framerate and the quality of the pointcloud. In principle, a higher framerate leads to less [...]

Thx, it helps a lot.

closed 2020-01-15 by Carolinadeveloper