ifm/ifm3d-ros #29: How to fix "[WARN]: Timeout waiting for camera! " after network interruption

Carolinadeveloper, 2020-01-12:
Hi guys,

when  the camera occurs network interruption, after I reconnecting the network,  **I can ping the sensor actually**, but the camera driver always throw out an warning "[WARN]: Timeout waiting for camera! "(shown as the following img) and not work anymore. My current solution is reboot the camera(i.e, power off and power on again), but I want to use this camera in some real product which may not allowed to take too much time on reboot. Is there anyway to fix this issue? 

[image]

theseankelly, 2020-01-14:
Hey @Carolinadeveloper -- following up. Could you share the configuration of your camera via `ifm3d dump`? I'm specifically interested in whether you're using free-running or software triggered mode. It'd be good to see the whole configuration though.

Carolinadeveloper, 2020-01-14:
> Hey @Carolinadeveloper -- following up. Could you share the configuration of your camera via `ifm3d dump`? I'm specifically interested in whether you're using free-running or software triggered mode. It'd be good to see the whole configuration though.

Hi, thank u for your reply, the following is my configuration:

Note: Since my project needs very high frame rate, I sped up the frame rate of pointcloud, I changed the following three parameters:

"ExposureTime" to "500"
"FrameRate" to "30"
"Type" to "under5m_low"

[image]
[image]
[image]

If u find something wrong with my configuration please contact me

Best regards,

Carolina

theseankelly, 2020-02-13:
FYI we're looking in to this.

theseankelly, 2020-02-17:
@grenait is taking a look at reproducing this issue.

lola-masson, 2022-06-28:
Closing due to inactivity. Please reopen is issue persists.

closed 2022-06-28 by lola-masson