ifm/ifm3d-ros #31: issue with starting camera node

NKKD, 2020-01-20:
~$ roslaunch ifm3d camera.launch

RLException: [camera.launch] is neither a launch file in package [ifm3d] nor is [ifm3d] a launch file name
The traceback for the exception was written to the log file

theseankelly, 2020-01-20:
Are you sourcing from an install space or devel?

From your ROS environment, can you please share the output of:
```
rospack find ifm3d
```

And then assuming it was found, please share the output of:
```
roscd ifm3d
ls launch
```

NKKD, 2020-01-20:
Thank you for replying @theseankelly 

~$ rospack find ifm3d
[rospack] Error: package 'ifm3d' not found

it seems like i didn't install my ifm3d properly.

can you please have a look on my bashrc. I'm currently using ros moledic.
I install my ros in /opt, ifm3d&ifm3d-ros in usr/include

if [ -f /opt/ros/melodic/setup.bash ]; then
  source /opt/ros/melodic/setup.bash
fi

cd ${opt}

export LPR_ROS=${opt}/ros

if [ -d ${LPR_ROS} ]; then
    for i in $(ls ${LPR_ROS}); do
        if [ -d ${LPR_ROS}/${i} ]; then
            if [ -f ${LPR_ROS}/${i}/setup.bash ]; then
                source ${LPR_ROS}/${i}/setup.bash --extend
            fi
        fi
    done
fi

NKKD, 2020-01-21:
I tried reinstall my ifm3d and ifm3d-ros. but it still gives me same error.

theseankelly, 2020-01-21:
Hi @NKKD -- the script you've pasted above (which I believe you found here: https://github.com/ifm/ifm3d-ros/blob/master/doc/building.md) is merely a convenience helper script. The idea is that if you `catkin_make install` all of your packages to a single location (say `~/ros`) this script will walk through the subfolders (presumably packages) found that directory and source the `setup.bash` script for each package it finds. So, the line `export LPR_ROS=${opt}/ros` should read `export LPR_ROS=path` where `path` is the fully qualified path to the parent folder containing your `ifm3d` installation directory.

This script is not mandatory and I'd suggest instead you start out sourcing the setup.bash script directly for the ifm3d-ros package to make sure the fundamentals are working: `source /path/to/ifm3d-ros/installation/directory/setup.bash` and then `rospack find ifm3d`

Also, FWIW (and not specific to ifm3d-ros): I'd suggest not installing to `/usr/include` as catkin packages contain full executables, not just headers. I tend to use my home directory or /opt is a good option as well.

NKKD, 2020-01-21:
Hey @theseankelly 
I tried to source the setup.bash file before run rospack find ifm3d but still doesn't work for me.

j@m15:~$ source /home/j/Documents/catkin/ifm3d/devel/setup.bash 
j@m15:~$ rospack find ifm3d
[rospack] Error: package 'ifm3d' not found

NKKD, 2020-01-21:
Here is my full installation history

j@m15:~$ cd /home/j/Documents/
j@m15:~/Documents$ ls
ifm3d-examples-master  ifm3d-pcl-viewer-master  o3d3xx-catkin
ifm3d-master           ifm3d-ros-catkin
j@m15:~/Documents$ git clone https://github.com/lovepark/ifm3d-ros.gitCloning into 'ifm3d-ros'...
remote: Enumerating objects: 305, done.
remote: Total 305 (delta 0), reused 0 (delta 0), pack-reused 305
Receiving objects: 100% (305/305), 1.61 MiB | 1.40 MiB/s, done.
Resolving deltas: 100% (160/160), done.
j@m15:~/Documents$ cd ifm3d-ros-catkin/
j@m15:~/Documents/ifm3d-ros-catkin$ mkdir ifm3d
j@m15:~/Documents/ifm3d-ros-catkin$ cd ifm3d
j@m15:~/Documents/ifm3d-ros-catkin/ifm3d$ mkdir src
j@m15:~/Documents/ifm3d-ros-catkin/ifm3d$ cd src
j@m15:~/Documents/ifm3d-ros-catkin/ifm3d/src$ catkin_init_workspace 
Creating symlink "/home/j/Documents/ifm3d-ros-catkin/ifm3d/src/CMakeLists.txt" pointing to "/opt/ros/melodic/share/catkin/cmake/toplevel.cmake"
j@m15:~/Documents/ifm3d-ros-catkin/ifm3d/src$ ln -s ~/home/j/Documents/ifm3d-ros ifm3d
j@m15:~/Documents/ifm3d-ros-catkin/ifm3d/src$ pwd
/home/j/Documents/ifm3d-ros-catkin/ifm3d/src
j@m15:~/Documents/ifm3d-ros-catkin/ifm3d/src$ ls -l
total 0
lrwxrwxrwx 1 j j 50 Jan 22 08:22 CMakeLists.txt -> /opt/ros/melodic/share/catkin/cmake/toplevel.cmake
lrwxrwxrwx 1 j j 34 Jan 22 08:22 ifm3d -> /home/j/home/j/Documents/ifm3d-ros
j@m15:~/Documents/ifm3d-ros-catkin/ifm3d/src$ cd ..
j@m15:~/Documents/ifm3d-ros-catkin/ifm3d$ catkin_make -DCATKIN_ENABLE_TESTING=ON
Base path: /home/j/Documents/ifm3d-ros-catkin/ifm3d
Source space: /home/j/Documents/ifm3d-ros-catkin/ifm3d/src
Build space: /home/j/Documents/ifm3d-ros-catkin/ifm3d/build
Devel space: /home/j/Documents/ifm3d-ros-catkin/ifm3d/devel
Install space: /home/j/Documents/ifm3d-ros-catkin/ifm3d/install
####
#### Running command: "cmake /home/j/Documents/ifm3d-ros-catkin/ifm3d/src -DCATKIN_ENABLE_TESTING=ON -DCATKIN_DEVEL_PREFIX=/home/j/Documents/ifm3d-ros-catkin/ifm3d/devel -DCMAKE_INSTALL_PREFIX=/home/j/Documents/ifm3d-ros-catkin/ifm3d/install -G Unix Makefiles" in "/home/j/Documents/ifm3d-ros-catkin/ifm3d/build"
####
-- The C compiler identification is GNU 7.4.0
-- The CXX compiler identification is GNU 7.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using CATKIN_DEVEL_PREFIX: /home/j/Documents/ifm3d-ros-catkin/ifm3d/devel
-- Using CMAKE_PREFIX_PATH: /ros/ifm3d;/opt/ros/melodic
-- This workspace overlays: /ros/ifm3d;/opt/ros/melodic
-- Found PythonInterp: /usr/bin/python2 (found suitable version "2.7.17", minimum required is "2") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python2
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/j/Documents/ifm3d-ros-catkin/ifm3d/build/test_results
-- Found gtest sources under '/usr/src/googletest': gtests will be built
-- Found gmock sources under '/usr/src/googletest': gmock will be built
-- Found PythonInterp: /usr/bin/python2 (found version "2.7.17") 
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
[... 62 lines cut ...]
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/j/Documents/ifm3d-ros-catkin/ifm3d/build/test_results
-- Found gtest sources under '/usr/src/googletest': gtests will be built
-- Found gmock sources under '/usr/src/googletest': gmock will be built
-- Found PythonInterp: /usr/bin/python2 (found version "2.7.17") 
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.20
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- Configuring done
-- Generating done
-- Build files have been written to: /home/j/Documents/ifm3d-ros-catkin/ifm3d/build
####
#### Running command: "make install -j12 -l12" in "/home/j/Documents/ifm3d-ros-catkin/ifm3d/build"
####
Install the project...
-- Install configuration: ""
-- Installing: /home/j/Documents/ifm3d/_setup_util.py
-- Installing: /home/j/Documents/ifm3d/env.sh
-- Installing: /home/j/Documents/ifm3d/setup.bash
-- Installing: /home/j/Documents/ifm3d/local_setup.bash
-- Installing: /home/j/Documents/ifm3d/setup.sh
-- Installing: /home/j/Documents/ifm3d/local_setup.sh
-- Installing: /home/j/Documents/ifm3d/setup.zsh
-- Installing: /home/j/Documents/ifm3d/local_setup.zsh

NKKD, 2020-01-21:
Thank you so much @theseankelly  I made it work. I found out the problem is that i forgot to delete a tilde in the path.

closed 2020-01-21 by NKKD

theseankelly, 2020-01-22:
@NKKD glad to hear it!