ifm/ifm3d-ros #32: Cannot build in Raspberry Pi 3B

pfontana96, 2020-01-27:
Hi everyone,
I'm not sure this is the right place for posting this question but I couldn't find many posts related to this package in [ROS Answers](https://answers.ros.org/questions/), please excuse me if you find this question out of place. I'm trying to install the ros package on a RPi 3B (Ubuntu-MATE 18.04 and ROS Melodic) by following [this](https://github.com/ifm/ifm3d-ros/blob/master/doc/building.md) instructions, I've already installed the ifm3d library but when I get to the step:

`catkin_make -DCATKIN_ENABLE_TESTING=ON`

The system freezes at 97% of the build (after some previous ctrl+C and reruns). If I run  `watch -n 0.5 free -m` I can see that almost all of my RAM is consumed. I've tried compiling with the -j1 flag but with no success (all the memory is also taken up). Do you have any idea how I can resolve it? I leave you here the output of my terminal (I had to interrupt it because it'd been hung there for nearly 30min):

```
Base path: /home/ebot/catkin/ifm3d
Source space: /home/ebot/catkin/ifm3d/src
Build space: /home/ebot/catkin/ifm3d/build
Devel space: /home/ebot/catkin/ifm3d/devel
Install space: /home/ebot/catkin/ifm3d/install
####
#### Running command: "cmake /home/ebot/catkin/ifm3d/src -DCATKIN_ENABLE_TESTING=ON -DCATKIN_DEVEL_PREFIX=/home/ebot/catkin/ifm3d/devel -DCMAKE_INSTALL_PREFIX=/home/ebot/catkin/ifm3d/install -G Unix Makefiles" in "/home/ebot/catkin/ifm3d/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/ebot/catkin/ifm3d/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/melodic
-- This workspace overlays: /opt/ros/melodic
-- Found PythonInterp: /usr/bin/python2 (found suitable version "2.7.17", minimum required is "2")
-- Using PYTHON_EXECUTABLE: /usr/bin/python2
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/ebot/catkin/ifm3d/build/test_results
-- Found gtest sources under '/usr/src/googletest': gtests will be built
-- Found gmock sources under '/usr/src/googletest': gmock will be built
-- Found PythonInterp: /usr/bin/python2 (found version "2.7.17")
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.20
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 1 packages in topological order:
-- ~~  - ifm3d
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'ifm3d'
-- ==> add_subdirectory(ifm3d)
-- ifm3d found component: camera
-- ifm3d found component: framegrabber
-- ifm3d found component: image
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- ifm3d: 1 messages, 6 services
-- Configuring done
-- Generating done
-- Build files have been written to: /home/ebot/catkin/ifm3d/build
####
#### Running command: "make -j4 -l4" in "/home/ebot/catkin/ifm3d/build"
####
[  0%] Built target _ifm3d_generate_messages_check_deps_Dump
[  0%] Built target _ifm3d_generate_messages_check_deps_SoftOff
[  0%] Built target _ifm3d_generate_messages_check_deps_Trigger
[  0%] Built target _ifm3d_generate_messages_check_deps_Extrinsics
[  0%] Built target std_msgs_generate_messages_cpp
[  0%] Built target _ifm3d_generate_messages_check_deps_SoftOn
[  0%] Built target std_msgs_generate_messages_py
[  0%] Built target _ifm3d_generate_messages_check_deps_SyncClocks
[  0%] Built target _ifm3d_generate_messages_check_deps_Config
[  0%] Built target std_msgs_generate_messages_eus
[  0%] Built target std_msgs_generate_messages_nodejs
[  0%] Built target std_msgs_generate_messages_lisp
[ 17%] Built target ifm3d_generate_messages_cpp
[ 35%] Built target ifm3d_generate_messages_nodejs
[ 57%] Built target ifm3d_generate_messages_py
[ 77%] Built target ifm3d_generate_messages_eus
[ 95%] Built target ifm3d_generate_messages_lisp
[ 95%] Built target ifm3d_generate_messages
[ 97%] Building CXX object ifm3d/CMakeFiles/ifm3d_ros.dir/src/camera_nodelet.cpp.o
^Cifm3d/CMakeFiles/ifm3d_ros.dir/build.make:62: recipe for target 'ifm3d/CMakeFiles/ifm3d_ros.dir/src/camera_nodelet.cpp.o' failed
make[2]: *** [ifm3d/CMakeFiles/ifm3d_ros.dir/src/camera_nodelet.cpp.o] Interrupt
Traceback (most recent call last):
  File "/opt/ros/melodic/bin/catkin_make", line 298, in <module>
    sys.exit(main())
  File "/opt/ros/melodic/bin/catkin_make", line 242, in main
    run_command(cmd, make_path)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/catkin/builder.py", line 240, in run_command
    proc.wait()
  File "/usr/lib/python2.7/subprocess.py", line 1099, in wait
    pid, sts = _eintr_retry_call(os.waitpid, self.pid, 0)
  File "/usr/lib/python2.7/subprocess.py", line 125, in _eintr_retry_call
    return func(*args)
KeyboardInterrupt
CMakeFiles/Makefile2:3777: recipe for target 'ifm3d/CMakeFiles/ifm3d_ros.dir/all' failed
make[1]: *** [ifm3d/CMakeFiles/ifm3d_ros.dir/all] Interrupt
Makefile:140: recipe for target 'all' failed
make: *** [all] Interrupt
```

Thank you in advance for your time!

theseankelly, 2020-01-27:
Hey @pfontana96 -- truth be told, I haven't actually tried installing ifm3d/ifm3d-ros on a Raspberry Pi. I can't say for sure if RAM is your issue in the above output, but I've definitely run into RAM issues compiling things like OpenCV before. The solution is to temporarily increase swap space. Here's one (of many) tutorials on how to do that:

https://www.bitpi.co/2015/02/11/how-to-change-raspberry-pis-swapfile-size-on-rasbian/

Let me know if that helps.

pfontana96, 2020-01-27:
@theseankelly  it was indeed a memory issue, I increased swap space and the package is working fine! Thank you for your quick response

closed 2020-01-27 by pfontana96