ifm/ifm3d-ros #36: how can get camera o3d303 intrinsic params

comwise, 2020-06-15:
i add the following code， but the data is null
`      //
      // publish intrinsics
      //
      ifm3d::Intrinsics intrinsics_msg;
      intrinsics_msg.header = optical_head;
      try
        {
          if(intrinsics.size() >= 16) {
            intrinsics_msg.fx       = intrinsics.at(0);
            intrinsics_msg.fy       = intrinsics.at(1);
            intrinsics_msg.mx       = intrinsics.at(2);
            intrinsics_msg.my       = intrinsics.at(3);
            intrinsics_msg.alpha    = intrinsics.at(4);
            intrinsics_msg.k1       = intrinsics.at(5);
            intrinsics_msg.k2       = intrinsics.at(6);
            intrinsics_msg.k3       = intrinsics.at(8);
            intrinsics_msg.k4       = intrinsics.at(9);
            intrinsics_msg.k5       = intrinsics.at(7);
            intrinsics_msg.trans_x  = intrinsics.at(10);
            intrinsics_msg.trans_y  = intrinsics.at(11);
            intrinsics_msg.trans_z  = intrinsics.at(12);
            intrinsics_msg.rot_x    = intrinsics.at(13);
            intrinsics_msg.rot_y    = intrinsics.at(14);
            intrinsics_msg.rot_z    = intrinsics.at(15);
          }
        }
      catch (const std::exception& ex)
        {
          ROS_WARN_STREAM("fetching intrinsics exception: " << ex.what());
        }
      this->intrinsics_pub_.publish(intrinsics_msg);` 

hardware firmware version is O3D3XX_Firmware_1.30.5309.swu

theseankelly, 2020-06-16:
Hey @comwise -- the `schema_mask` parameter on the node determines which information is transmit from the camera. By default, this parameter is '10', which corresponds to amplitude, XYZ, confidence, and extrinsics.

Intrinsic calibration is specified by bit 8 (0x100). Please set the `schema_mask` parameter when launching the ros node to 266 (decimal) to enable the intrinsic calibration data as well.

i.e:
```
roslaunch ifm3d camera.launch schema_mask:=266
```

comwise, 2020-06-17:
@theseankelly thanks very much， i get the intrinsic params now

theseankelly, 2020-06-17:
Sure, feel free to reach out if you have any further issues.

closed 2020-06-17 by theseankelly