ifm/ifm3d-ros #37: v0.18.0 ifm3d Compatibility

cartesian-theatrics, 2020-06-29:
Hello, 

v0.6.2 does not seem to be compatible with v0.18.0 of `ifm3d`. Is this the case? Any estimate of when we can expect a compatible release?

theseankelly, 2020-06-29:
It's compatible -- and tested on kinetic, melodic, and noetic. I've just forgotten to update the README.

BTW we're a part of the ROS distribution for those three distros now -- `$ sudo apt install ros-<distro>-ifm3d`

cartesian-theatrics, 2020-06-29:
I'm getting this now when building ifm3d-ros after upgrading the versions:
```
CMake Error at catkin_ws/src/ifm3d-ros/CMakeLists.txt:71 (add_library):
  Target "ifm3d_ros" links to target "Boost::system" but the target was not
  found.  Perhaps a find_package() call is missing for an IMPORTED target, or
  an ALIAS target is missing?

CMake Error at catkin_ws/src/ifm3d-ros/CMakeLists.txt:71 (add_library):
  Target "ifm3d_ros" links to target "Threads::Threads" but the target was
  not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?
```
it appears as tho It's not available as a link target in ifm3d anymore.

theseankelly, 2020-06-30:
What OS and ROS distro are you targeting?

How are you installing ifm3d (building from source, from the apt repository, other?)

cartesian-theatrics, 2020-07-15:
This is with Kinetic. The issue ended up being an error in our build. How I'm dealing what appears to be an incorrect version of xmlrpc-c++ in ros:
```
[ERROR] [1594800006.741875968]: Failed to load nodelet [/camera] of type [ifm3d_ros/camera_nodelet] even after refreshing the cache: Failed to load library /opt/catkin/lib//libifm3d_ros.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = /opt/catkin/lib//libifm3d_ros.so: undefined symbol: _ZTIN8xmlrpc_c18carriageParm_http0E)
```

I made sure that the ubuntu build dependencies are there:
```
```
$ sudo apt-get install libxmlrpc-c++8-dev
$ sudo apt-get install libgoogle-glog-dev
```

theseankelly, 2020-07-15:
Are you compiling the core `ifm3d` library from source or using apt? If apt, are you using the ifm-provided apt mirror (from the README) or are you using the ROS distribution version (package `ros-kinetic-ifm3d-core`)?

Same question for the `ifm3d-ros` package -- are you building the catkin package from source or are you using the ROS distribution (package `ros-kinetic-ifm3d`)?

cartesian-theatrics, 2020-07-15:
Both are compiled from source.

theseankelly, 2020-07-16:
Strange. I wonder if `ifm3d` is linking against a system library when it should be linking against a ros-provided version of the same library. Are you compiling `ifm3d` from a ROS environment (aka after souring `/opt/ros/kinetic/setup.bash`)?

I can try to reproduce -- would you be able to provide the output of `rospack list` so I can make sure I've got as similar an environment as possible?

theseankelly, 2020-07-16:
And also the output of `ldd /opt/catkin/lib/libifm3d_ros.so`?

cartesian-theatrics, 2020-07-17:
Everything is built from source, including `ifm3d` and `ifm3d-ros`. There may be something from from my local environment leaking into the build environment. `libifm3d_ros.so` is built by the `ifm3d-ros` library correct?

ldd shows it's loading the libxmlrpcpp.so in ros.
```
└─▪ ldd /opt/catkin/lib//libifm3d_ros.so
	linux-vdso.so.1 =>  (0x00007ffcf753a000)
	libimage_transport.so => /opt/ros/kinetic/lib/libimage_transport.so (0x00007fbd08000000)
	libcv_bridge.so => /opt/ros/kinetic/lib/libcv_bridge.so (0x00007fbd07dd6000)
	libopencv_core3.so.3.3 => /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_core3.so.3.3 (0x00007fbd06e9d000)
	libnodeletlib.so => /opt/ros/kinetic/lib/libnodeletlib.so (0x00007fbd06c36000)
	libclass_loader.so => /opt/ros/kinetic/lib/libclass_loader.so (0x00007fbd06a12000)
	libroscpp.so => /opt/ros/kinetic/lib/libroscpp.so (0x00007fbd06676000)
	librosconsole.so => /opt/ros/kinetic/lib/librosconsole.so (0x00007fbd06444000)
	libroscpp_serialization.so => /opt/ros/kinetic/lib/libroscpp_serialization.so (0x00007fbd06241000)
	librostime.so => /opt/ros/kinetic/lib/librostime.so (0x00007fbd06014000)
	libconsole_bridge.so.0.2 => /usr/lib/x86_64-linux-gnu/libconsole_bridge.so.0.2 (0x00007fbd05e0f000)
	libboost_system.so.1.58.0 => /usr/lib/x86_64-linux-gnu/libboost_system.so.1.58.0 (0x00007fbd05c0b000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007fbd059ee000)
	libglog.so.0 => /usr/lib/x86_64-linux-gnu/libglog.so.0 (0x00007fbd057bf000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007fbd0543d000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007fbd05134000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007fbd04f1e000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fbd04b54000)
	/lib64/ld-linux-x86-64.so.2 (0x00007fbd08586000)
	libmessage_filters.so => /opt/ros/kinetic/lib/libmessage_filters.so (0x00007fbd0494f000)
	libtinyxml2.so.2 => /usr/lib/x86_64-linux-gnu/libtinyxml2.so.2 (0x00007fbd0473c000)
	libroslib.so => /opt/ros/kinetic/lib/libroslib.so (0x00007fbd0452a000)
	libboost_filesystem.so.1.58.0 => /usr/lib/x86_64-linux-gnu/libboost_filesystem.so.1.58.0 (0x00007fbd04312000)
	libopencv_imgcodecs3.so.3.3 => /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_imgcodecs3.so.3.3 (0x00007fbd03ce3000)
	libboost_regex.so.1.58.0 => /usr/lib/x86_64-linux-gnu/libboost_regex.so.1.58.0 (0x00007fbd039db000)
	libopencv_imgproc3.so.3.3 => /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_imgproc3.so.3.3 (0x00007fbd010e5000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007fbd00ee1000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007fbd00cd9000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007fbd00abf000)
	libbondcpp.so => /opt/ros/kinetic/lib/libbondcpp.so (0x00007fbd0088f000)
	libboost_thread.so.1.58.0 => /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.58.0 (0x00007fbd00669000)
	libPocoFoundation.so.9 => /usr/lib/libPocoFoundation.so.9 (0x00007fbd0030d000)
	libxmlrpcpp.so => /opt/ros/kinetic/lib/libxmlrpcpp.so (0x00007fbd000ee000)
	libcpp_common.so => /opt/ros/kinetic/lib/libcpp_common.so (0x00007fbcffee6000)
	libboost_chrono.so.1.58.0 => /usr/lib/x86_64-linux-gnu/libboost_chrono.so.1.58.0 (0x00007fbcffcde000)
	librosconsole_log4cxx.so => /opt/ros/kinetic/lib/librosconsole_log4cxx.so (0x00007fbcffac5000)
	librosconsole_backend_interface.so => /opt/ros/kinetic/lib/librosconsole_backend_interface.so (0x00007fbcff8c3000)
	liblog4cxx.so.10 => /usr/lib/x86_64-linux-gnu/liblog4cxx.so.10 (0x00007fbcff4ec000)
	libgflags.so.2 => /usr/lib/x86_64-linux-gnu/libgflags.so.2 (0x00007fbcff2cb000)
	libunwind.so.8 => /usr/lib/x86_64-linux-gnu/libunwind.so.8 (0x00007fbcff0b0000)
	librospack.so => /opt/ros/kinetic/lib/librospack.so (0x00007fbcfee6e000)
	libjpeg.so.8 => /usr/lib/x86_64-linux-gnu/libjpeg.so.8 (0x00007fbcfec15000)
	libwebp.so.5 => /usr/lib/x86_64-linux-gnu/libwebp.so.5 (0x00007fbcfe9b9000)
	libpng12.so.0 => /lib/x86_64-linux-gnu/libpng12.so.0 (0x00007fbcfe794000)
	libtiff.so.5 => /usr/lib/x86_64-linux-gnu/libtiff.so.5 (0x00007fbcfe51f000)
	libjasper.so.1 => /usr/lib/x86_64-linux-gnu/libjasper.so.1 (0x00007fbcfe2ca000)
	libicui18n.so.55 => /usr/lib/x86_64-linux-gnu/libicui18n.so.55 (0x00007fbcfde68000)
	libicuuc.so.55 => /usr/lib/x86_64-linux-gnu/libicuuc.so.55 (0x00007fbcfdad4000)
	libuuid.so.1 => /lib/x86_64-linux-gnu/libuuid.so.1 (0x00007fbcfd8cf000)
	libpcre.so.3 => /lib/x86_64-linux-gnu/libpcre.so.3 (0x00007fbcfd65f000)
	libapr-1.so.0 => /usr/lib/x86_64-linux-gnu/libapr-1.so.0 (0x00007fbcfd42d000)
	libaprutil-1.so.0 => /usr/lib/x86_64-linux-gnu/libaprutil-1.so.0 (0x00007fbcfd206000)
	liblzma.so.5 => /lib/x86_64-linux-gnu/liblzma.so.5 (0x00007fbcfcfe4000)
	libtinyxml.so.2.6.2 => /usr/lib/x86_64-linux-gnu/libtinyxml.so.2.6.2 (0x00007fbcfcdce000)
	libboost_program_options.so.1.58.0 => /usr/lib/x86_64-linux-gnu/libboost_program_options.so.1.58.0 (0x00007fbcfcb50000)
	libpython2.7.so.1.0 => /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0 (0x00007fbcfc5c2000)
	libjbig.so.0 => /usr/lib/x86_64-linux-gnu/libjbig.so.0 (0x00007fbcfc3b4000)
	libicudata.so.55 => /usr/lib/x86_64-linux-gnu/libicudata.so.55 (0x00007fbcfa8fd000)
	libcrypt.so.1 => /lib/x86_64-linux-gnu/libcrypt.so.1 (0x00007fbcfa6c5000)
	libexpat.so.1 => /lib/x86_64-linux-gnu/libexpat.so.1 (0x00007fbcfa49c000)
	libutil.so.1 => /lib/x86_64-linux-gnu/libutil.so.1 (0x00007fbcfa299000)
```

theseankelly, 2020-07-30:
Hey @cartesian-theatrics sorry for my terrible response time on this topic.

Yeah, libifm3d_ros.so is from the `ifm3d-ros` package.

I need to get set up for a repro on Kinetc -- can you confirm how you're launching the nodelet (via launch or directly via a manager?)

cartesian-theatrics, 2020-07-30:
No worries, the problem ended being that ifm3d was being statically linked. Once I enabled building of shared libs it worked.

theseankelly, 2020-07-30:
Weird. `ifm3d` should be building shared libraries by default. Was that something you explicitly configured or do we have a CMake bug?

closed 2020-08-03 by theseankelly