ifm/ifm3d-ros #42: O3DP03 Topic name of pallet detection data

ZhihangCui, 2021-11-12:
Using O3DP03 to detect pallet with command：
`roslaunch ifm3d camera.launch
roslaunch ifm3d rviz.launch`
however there is no topic about pallet pose
`~$ rostopic list
/clicked_point
/ifm3d/camera/amplitude
/ifm3d/camera/amplitude/compressed
/ifm3d/camera/amplitude/compressed/parameter_descriptions
/ifm3d/camera/amplitude/compressed/parameter_updates
/ifm3d/camera/cloud
/ifm3d/camera/confidence
/ifm3d/camera/confidence/compressed
/ifm3d/camera/confidence/compressed/parameter_descriptions
/ifm3d/camera/confidence/compressed/parameter_updates
/ifm3d/camera/distance
/ifm3d/camera/distance/compressed
/ifm3d/camera/distance/compressed/parameter_descriptions
/ifm3d/camera/distance/compressed/parameter_updates
/ifm3d/camera/extrinsics
/ifm3d/camera/good_bad_pixels
/ifm3d/camera/good_bad_pixels/compressed
/ifm3d/camera/good_bad_pixels/compressed/parameter_descriptions
/ifm3d/camera/good_bad_pixels/compressed/parameter_updates
/ifm3d/camera/raw_amplitude
/ifm3d/camera/raw_amplitude/compressed
/ifm3d/camera/raw_amplitude/compressed/parameter_descriptions
/ifm3d/camera/raw_amplitude/compressed/parameter_updates
/ifm3d/camera/unit_vectors
/ifm3d/camera/xyz_image
/ifm3d/camera/xyz_image/compressed
/ifm3d/camera/xyz_image/compressed/parameter_descriptions
/ifm3d/camera/xyz_image/compressed/parameter_updates
/ifm3d/camera_standalone_nodelet/bond
/initialpose
/move_base_simple/goal
/rosout
/rosout_agg
/tf
/tf_static
`
Topic /move_base_simple/goal isn't from O3DP03 ,  no data in it.
Did I use the wrong parameter?

lola-masson, 2021-11-12:
@ZhihangCui it seems to me that you are using the wrong ROS tool. You should've received a software package along with the O3DPO3, is that what you are using?

ZhihangCui, 2021-11-15:
> @ZhihangCui it seems to me that you are using the wrong ROS tool. You should've received a software package along with the O3DPO3, is that what you are using?

@lola-masson Actually, I haven't received the software package for O3DPO3, could you tell me where to get it. Thanks !

lola-masson, 2021-11-15:
My bad, I wasn't clear. The PDS software should be installed on your O3DPO3. You can interact with it with the ifm vision assistant or through the PDS wire protocol. I expect that documentation should've been provided to you by your responsible sales engineer, if not I would recommend getting in touch with them.

lola-masson, 2022-06-28:
Closing due to inactivity. Please reopen if issue persists.

closed 2022-06-28 by lola-masson