ifm/ifm3d-ros #47: Cmake error

hwil003, 2022-03-11:
I am getting this error when I try to catkin_make the workspace:
CMake Error at ifm3d-ros/ifm3d_ros_driver/CMakeLists.txt:4 (find_package):
  By not providing "Findifm3d.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "ifm3d", but
  CMake did not find one.

  Could not find a package configuration file provided by "ifm3d" with any of
  the following names:

    ifm3dConfig.cmake
    ifm3d-config.cmake

  Add the installation prefix of "ifm3d" to CMAKE_PREFIX_PATH or set
  "ifm3d_DIR" to a directory containing one of the above files.  If "ifm3d"
  provides a separate development package or SDK, be sure it has been
  installed.

-- Configuring incomplete, errors occurred!
See also "/home/hanson/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/hanson/catkin_ws/build/CMakeFiles/CMakeError.log".
make: *** [Makefile:320: cmake_check_build_system] Error 1
Invoking "make cmake_check_build_system" failed

desengph, 2022-03-11:
Hi @hwil003,

can you please give us more information about what ifm camera you are using (e.g., O3D3xx, O3X, O3R) and the setup procedure you have gone through so far?

I'm guessing that you are trying to build the ROS node for a O3R camera, since you are using our latest implementation. 
Please see the build instructions here: [ifm3d-ros building instructions](https://github.com/ifm/ifm3d-ros/blob/master/ifm3d_ros_driver/doc/noetic.md)

If you need further information about the ifm3d build process, see this documentation: [ifm3d linux source build](http://ifm3d.com/sphinx-doc/build/html/ifm3d/doc/sphinx/content/installation_instructions/source_build.html)

If you are planning to build the ifm3d-ros node for any other ifm camera than the O3R please have a look at the tagged releases here: [ifm3d-ros v0.7.0](https://github.com/ifm/ifm3d-ros/releases/tag/v0.7.0).

hwil003, 2022-03-12:
Hey Phillip,
I am using the O3X100 camera. Is there a different setup for that?

Regards
Hanson

hwil003, 2022-03-12:
My setp up procedure has been from this instruction:
https://github.com/ifm/ifm3d-ros/blob/master/ifm3d_ros_driver/doc/building.md
But for step 3 I didn't have the CMake list in my ifm3d-ros folder. And so
when I did step 4 the catkin_make failed.

hwil003, 2022-03-15:
> Hi @hwil003 <https://github.com/hwil003>,
> please follow the instructions in the tagged release version 0.7.0
> The master branch is a work in progress and can only be used with the
> latest ifm camera: O3R.
>
> The release includes a complete mirror of the repository. You'll find the
> build instructions and README in there.

Ok thanks I will check this out

hwil003, 2022-03-18:
Hey Phillip, I'm stilling getting this error when I try to catkin_make:
CMake Error at ifm3d-ros-0.7.0/CMakeLists.txt:4 (find_package):
  Could not find a package configuration file provided by "ifm3d" (requested
  version 0.19.0) with any of the following names:

    ifm3dConfig.cmake
    ifm3d-config.cmake

  Add the installation prefix of "ifm3d" to CMAKE_PREFIX_PATH or set
  "ifm3d_DIR" to a directory containing one of the above files.  If "ifm3d"
  provides a separate development package or SDK, be sure it has been
  installed.

-- Configuring incomplete, errors occurred!
See also "/home/hanson/3d_camera_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/hanson/3d_camera_ws/build/CMakeFiles/CMakeError.log".
make: *** [Makefile:320: cmake_check_build_system] Error 1
Invoking "make cmake_check_build_system" failed

The 3d camera is plugged in and everything. Is there something wrong with
the package?

desengph, 2022-03-21:
Hi @hwil003,
can you please double-check that the ifm3d package is installed on your machine and accessible to cmake.

If you use the standard build procedure, the package will be “installed” under the path `/usr/lib`
Please see the docmentation for installing the ifm3d C++ API [here](https://github.com/ifm/ifm3d/blob/master/doc/source_build.md#the-default-build).
The ifm3d C++ API is a requirement for the ROS wrapper. Without the ifm3d C++ API it can not work / compile it.

If a Dockerfile / Docker image (ifm3d and ifm3d-ros wrapper) would help you get started for a quick testing, please let me know.

dekhanra, 2023-06-22:
@hwil003,

As there have been no recent updates to this GitHub issue, we're closing it.

If you still have questions, please feel free to open a new issue thread.

closed 2024-06-21 by desengph