ifm/ifm3d-ros #48: Gazebo plugin

dharapatel1596, 2022-03-24:
Hello,

I have one question about simulating my ifm O3D300 into gazebo? Is there any package available for it? Because I want to design my process with the robot and camera mounted on top (without actual robot).

Any leads will be appriciated.

TimFinkler, 2022-04-26:
@lola-masson  Thanks for your answer.
Is there any chance that you will develop this plugin soon?
It would be much appreciated 👍

lola-masson, 2022-04-26:
This is not in our list of priorities right now as we are working on the launch of our new platform. When things calm down, let's say in a couple months, we might consider it I think this is a good idea.

dharapatel1596, 2022-04-27:
Looking forward for it!

closed 2022-06-28 by lola-masson

dharapatel1596, 2022-06-28:
Hii @lola-masson, 
silly question, but where can I see this plugin in this repo or clone it? I have check the Master branch but I couldn't find it. 
Thanks in advance :)

lola-masson, 2022-06-28:
Hi @dharapatel1596, we do not yet have a gazebo plugin.