ifm/ifm3d-ros #52: camera's links relative to the housing

fmauch, 2022-07-27:
Hi,

we are using four O3Rs on our mobile robot and we seem to get along so far. However, I'm having trouble properly integrating the cameras into our URDF. From what I can see, pointclouds are published in `<base>_link`, while all images are published in `<base>_optical_link`.

In the provided launchfiles you have static tf publishers that publish from a common reference frame to different cameras' `_link` frames, but those are rather hard to estimate when you don't know where the link frame is supposed to be in relation to the camera's housing. Am I missing something or could you maybe provide the missing information? Maybe a description package would be rather useful so people could include that into their descriptions directly. As a starting point (maybe also for others) I currently use the following xacro macro:

```xml
  <xacro:macro name="ifm_base" params="*origin parent name">
    <link name="${name}_base_link">
      <visual>
        <origin xyz="0 0 0" rpy="0 0 0"/>
        <geometry>
          <mesh filename="package://my_description_package/meshes/visual/my_custom_mesh_with_housing.dae"
                scale="1 1 1"/>
        </geometry>
      </visual>
      <collision>
        <origin xyz="0 0 0" rpy="0 0 0"/>
        <geometry>
          <mesh filename="package://my_description_package/meshes/collision/my_custom_mesh_with_housing.stl"
                scale="1 1 1"/>
        </geometry>
      </collision>
    </link>
    <joint name="${name}_joint" type="fixed">
      <parent link="${parent}" />
      <child link = "${name}_base_link" />
      <xacro:insert_block name="origin" />
    </joint>
    <link name="${name}_link">
    </link>
    <joint name="${name}_camera_joint" type="fixed">
      <parent link="${name}_base_link" />
      <child link = "${name}_link" />
      <!-- where is the _link relative to the body? -->
      <origin xyz="0 0 0" rpy="0 0 0" />
    </joint>
    <link name="${name}_optical_link">
    </link>
    <joint name="${name}_camera_optical_joint" type="fixed">
      <parent link="${name}_link" />
      <child link = "${name}_optical_link" />
      <!-- where is the optical_link relative to _link? -->
      <origin xyz="0 0 0" rpy="-${pi/2} 0 0" />
    </joint>
  </xacro:macro>
```

As a note: We are using ifm3d_ros v1.0.1.

lola-masson, 2022-07-28:
Hi @fmauch,

You are right, it is not clear where the camera coordinate center is. It is at the center of the back plate, between the four screws. We will work on adding this to the doc, thanks for your feedback.

fmauch, 2022-07-28:
Thank you for your fast reply, that helped me a lot

fmauch, 2022-07-28:
In case we'd create a description package - would you be interested in a PR?

lola-masson, 2022-07-28:
Absolutely!

dekhanra, 2023-06-22:
@fmauch,
As there have been no recent updates to this GitHub issue, we're closing it.
If you still have questions, please feel free to open a new issue thread.

closed 2024-06-21 by desengph