ifm/ifm3d-ros #53: Compiling error/ camera_base.h

brio135, 2022-07-28:
Hi,
I am quite new on ros working, sorry if my questin seem silly. I installed the ifm3d libraries in my _usr/include/_ folder but when compiling one of my scritps that include _ifm3d_ros_msgs/camera_nodelet.h_ the file _camera_base.h_ seem to be missing. Cheking in the _ifm3d/camera_ folder the file is indeed not present. Where can I find this file?

lola-masson, 2022-07-28:
You might have installed the wrong ifm3d version. Which version do you have? You can check with `ifm3d --version`. The current version of `ifm3d-ros` requires ifm3d 0.93.0.

brio135, 2022-07-28:
Thank you for the answer. I think that my problem is different, I'm talking about this _https://github.com/ifm/ifm3d_, and the latest version is 0.18, and I have that. What am I missing?

lola-masson, 2022-07-28:
The latest version is actually [0.93.0](https://github.com/ifm/ifm3d/releases/tag/v0.93.0), it is only a "tagged" version and not a "released" version as it does not support all the ifm devices, but this is the version you should use for the O3R platform.

Gabiriyal, 2023-06-16:
i am aslo got same error like this but i am using ifm3d version as 1.2.6 and ifm-ros package as 1.0.1     { In file included from /home/ifm_ros/src/ifm3d-ros-1.0.1/ifm3d_ros_driver/src/camera_nodelet.cpp:6:
/home/ifm_ros/src/ifm3d-ros-1.0.1/ifm3d_ros_driver/include/ifm3d_ros_driver/camera_nodelet.h:18:10: fatal error: ifm3d/camera/camera_base.h: No such file or directory
   18 | #include <ifm3d/camera/camera_base.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [ifm3d-ros-1.0.1/ifm3d_ros_driver/CMakeFiles/ifm3d_ros.dir/build.make:63: ifm3d-ros-1.0.1/ifm3d_ros_driver/CMakeFiles/ifm3d_ros.dir/src/camera_nodelet.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:2560: ifm3d-ros-1.0.1/ifm3d_ros_driver/CMakeFiles/ifm3d_ros.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
Invoking "make -j8 -l8" failed

Gabiriyal, 2023-06-16:
please give me a solution

dekhanra, 2023-06-16:
Hi @Gabiriyal, please check the compatibility matrix for ifm3d_ros https://github.com/ifm/ifm3d-ros/tree/master#software-compatibility-matrix

Gabiriyal, 2023-06-16:
in that ifm-ros version is 1.0.0 and ifm3d is 0.93.0 
i got the ifm-ros package but i uable to get the 0.93.0 version of if3d 

and  i follow the instruction in the following website : https://ifm3d.com/index.html

Gabiriyal, 2023-06-16:
/ifm3d-0.91.0/build$ cmake -DCMAKE_INSTALL_PREFIX=/usr ..
-- Setting build type to 'RelWithDebInfo' as none was specified.
fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
CMake Error at cmake/modules/GetVersionFromGitTag.cmake:59 (string):
  string sub-command REGEX, mode MATCHALL needs at least 5 arguments total to
  command.
Call Stack (most recent call first):
  CMakeLists.txt:13 (include)

CMake Error at CMakeLists.txt:25 (project):
  VERSION "NOTFOUND.NOTFOUND.NOTFOUND" format invalid.

-- Configuring incomplete, errors occurred!

dekhanra, 2023-06-16:
@Gabiriyal,

Can you give bit more details on your software and hardware setup? What is the firmware installed on VPU and Ubuntu distribution?

lola-masson, 2023-06-16:
@Gabiriyal this is a known issue building from zip for some versions of ifm3d. Are you able to get the code from git? Clone the ifm3d repository, checkout the v0.93.0 branch and follow the source build instructions here: https://github.com/ifm/ifm3d/blob/v0.93.0/doc/sphinx/content/source_build.md.

Note that we have some pre-relase ifm3d-ros code that supports ifm3d 1.2.6, in this branch: https://github.com/ifm/ifm3d-ros/tree/dev1.2. It is not yet a fully tested version, but feel free to give it a try if you want to use more recent versions of ifm3d.

Gabiriyal, 2023-06-16:
i cannot install any firmware but in the website is there (OVP800 Firmware 1.0.14-973
OVP800 Firmware Version 1.0.14-973
) whether i need to install the firmware ,it is the latest hardware inbuild it has a current firmware are not? and my ubuntu distribution is noetic 
and also i cannot get the ifm3d version as a git repositry thanks for that .
Is that  there is  any deb files are need to install for the ifm3d-0.93.0

Gabiriyal, 2023-06-16:
ok thank u .i wiil check it

Gabiriyal, 2023-06-16:
one issue it shows 
~/ifm_ros/src$ git clone  https://github.com/ifm/ifm3d-ros/tree/dev1.2.
Cloning into 'dev1.2.'...
fatal: repository 'https://github.com/ifm/ifm3d-ros/tree/dev1.2./' not found

Gabiriyal, 2023-06-16:
into this link 'https://github.com/ifm/ifm3d-ros/tree/dev1.2. there is an repositry but it is an older version https://github.com/ifm/ifm3d-ros.git

Gabiriyal, 2023-06-16:
it shows an error 
~/ifm_ros/src$ git clone  https://github.com/ifm/ifm3d-ros/tree/dev1.2.
Cloning into 'dev1.2.'...
fatal: repository 'https://github.com/ifm/ifm3d-ros/tree/dev1.2./' not found
while enter into this  https://github.com/ifm/ifm3d-ros/tree/dev1.2 ,it having the older version of another git repositry

lola-masson, 2023-06-16:
Hey @Gabiriyal, you need to check out the branch dev1.2. Maybe you would find useful to learn about git usage, for example with: https://www.atlassian.com/git/tutorials/setting-up-a-repository/git-clone

Gabiriyal, 2023-06-17:
[FATAL] [1686974423.733924155]: Failed to load nodelet '/ifm3d_ros_driver/camera` of type `ifm3d_ros/camera_nodelet` to manager `camera_standalone_nodelet'
[ifm3d_ros_driver/camera-2] process has died [pid 5221, exit code 255, cmd /opt/ros/noetic/lib/nodelet/nodelet load ifm3d_ros/camera_nodelet camera_standalone_nodelet __name:=camera __log:=/home/robis/.ros/log/f22fe0fc-0cc2-11ee-837a-f9182d4e3521/ifm3d_ros_driver-camera-2.log].
log file: /home/robis/.ros/log/f22fe0fc-0cc2-11ee-837a-f9182d4e3521/ifm3d_ros_driver-camera-2*.log

i got error like this will using this command roslaunch ifm3d_ros_examples camera_2d.launch

Gabiriyal, 2023-06-17:
[ INFO] [1686975141.666864385]: Initializing camera...
[ INFO] [1686975146.677230126]: Initializing framegrabber...
[ INFO] [1686975151.677437140]: Start streaming frames
[ WARN] [1686975156.678626410]: Timeout waiting for camera!
[ WARN] [1686975156.678666220]: Attempting to restart framegrabber...
[ INFO] [1686975156.678681432]: Running dtors...
[ INFO] [1686975156.678854796]: Initializing camera...
[ INFO] [1686975161.691967945]: Initializing framegrabber...
[ INFO] [1686975166.692378127]: Start streaming frames
[ WARN] [1686975171.693039815]: Timeout waiting for camera!
[ WARN] [1686975171.693129855]: Attempting to restart framegrabber...
[ INFO] [1686975171.693171770]: Running dtors...
[ INFO] [1686975171.693577726]: Initializing camera...
[ INFO] [1686975176.705947934]: Initializing framegrabber...
[ INFO] [1686975181.706168308]: Start streaming frames
[ WARN] [1686975186.707217287]: Timeout waiting for camera!
[ WARN] [1686975186.707316279]: Attempting to restart framegrabber...
[ INFO] [1686975186.707379356]: Running dtors...
[ INFO] [1686975186.707818487]: Initializing camera...
[ INFO] [1686975191.722609885]: Initializing framegrabber...

i got time out error

Gabiriyal, 2023-06-17:
i got current error as time out erroe and i fix the nodelet process died

Gabiriyal, 2023-06-17:
@lola-masson give me the solution for that issue.

Gabiriyal, 2023-06-17:
@lola-masson I also change the time sync (time delay 1000ms into 10000 ms ) is it correct or not 

And there is another possibities to read the data from o3r camera using ros..
Can i use this https://github.com/ifm/ifm3d-ros/tree/master#software-compatibility-matrix 
But i need deb files for ifm3d 0.93.0bversion can you help for that..

dekhanra, 2023-06-19:
Hi @Gabiriyal, As you see in the [launch file](https://github.com/ifm/ifm3d-ros/blob/379d0443f56f7a0d9380e01b501a1f07b077b038/ifm3d_ros_examples/launch/camera_2d.launch#L8) the default pcic port is 50010.  
```bash
$roslaunch ifm3d_ros_examples camera_2d.launch pcic_port:=500XX
```
If it does not solve the issue please send the json configuration.

Gabiriyal, 2023-06-19:
i check the all pcic port 5001(1to5) but it shows as..same
and the dump.json configuration 
{
    "device": {
      "clock": {
        "currentTime": 1581092244602229792
      },
      "diagnostic": {
        "temperatures": [],
        "upTime": 1599000000000
      },
      "info": {
        "device": "0301",
        "deviceTreeBinaryBlob": "tegra186-quill-p3310-1000-c03-00-base.dtb",
        "features": {},
        "name": "",
        "partNumber": "M03903",
        "productionState": "AA",
        "serialNumber": "000201233338",
        "vendor": "0001"
      },
      "network": {
        "authorized_keys": "",
        "ipAddressConfig": 0,
        "macEth0": "00:04:4B:E4:DA:9E",
        "macEth1": "00:02:01:23:33:38",
        "networkSpeed": 1000,
        "staticIPv4Address": "192.168.0.69",
        "staticIPv4Gateway": "192.168.0.201",
        "staticIPv4SubNetMask": "255.255.255.0",
        "useDHCP": false
      },
      "state": {
        "errorMessage": "",
        "errorNumber": ""
      },
      "swVersion": {
        "kernel": "4.9.140-l4t-r32.4+g231628d3fa15",
        "l4t": "r32.4.3",
        "os": "0.13.3-188",
        "schema": "feature/O3R-4949",
        "swu": "0.15.3"
      }
    },
    "ports": {
      "port0": {
        "acquisition": {
          "exposureLong": 5000,
          "exposureShort": 400,
          "framerate": 10.0,
          "offset": 0.0,
          "version": {
            "major": 0,
            "minor": 0,
            "patch": 0
          }
        },
        "data": {
          "algoDebugConfig": {},
          "availablePCICOutput": [
            "AMPLITUDE_COMPRESSED",
            "CONFIDENCE",
            "RADIAL_DISTANCE_COMPRESSED",
            "RADIAL_DISTANCE_NOISE",
            "REFLECTIVITY",
            "TOF_INFO"
          ],
          "pcicTCPPort": 50010
        },
        "info": {
[... 245 lines cut ...]
          "sensor": "OV9782",
          "sensorID": "OV9782_127x80_noIllu_Csample",
          "serialNumber": "000000000094",
          "vendor": "0001"
        },
        "mode": "experimental_autoexposure2D",
        "processing": {
          "extrinsicHeadToUser": {
            "rotX": 0.0,
            "rotY": 0.0,
            "rotZ": 0.0,
            "transX": 0.0,
            "transY": 0.0,
            "transZ": 0.0
          },
          "version": {
            "major": 0,
            "minor": 0,
            "patch": 0
          }
        },
        "state": "CONF"
      }
    }
  }

Gabiriyal, 2023-06-19:
i using this ifm3d-ros package :  https://github.com/ifm/ifm3d-ros/tree/dev1.2

dekhanra, 2023-06-19:
From Your config file I can  see the setup as

PORT0: 3D --> RUN
PORT1: 3D --> RUN
PORT2: 2D --> CONF
PORT5: 2D --> CONF

If you are trying to launch 3D node then execute 
```bash
roslaunch ifm3d_ros_example camera_3d.launch pcic_port:=50011
```

if you are trying to launch 2D node then execute
```bash
roslaunch ifm3d_ros_example camera_2d.launch pcic_port:=50012
```

dekhanra, 2023-06-19:
Please change the PORT state to run before launching the camera node

Gabiriyal, 2023-06-19:
i also change the port state to run 

  "pcicTCPPort": 50015
        },
        "info": {
          "device": "2301",
          "deviceTreeBinaryBlobOverlay": "001-ov9782.dtbo",
          "features": {
            "fov": {
              "horizontal": 80,
              "vertical": 127
            },
            "resolution": {
              "height": 800,
              "width": 1280
            },
            "type": "2D"
          },
          "name": "",
          "partNumber": "M03934",
          "productionState": "AA",
          "sensor": "OV9782",
          "sensorID": "OV9782_127x80_noIllu_Csample",
          "serialNumber": "000000000094",
          "vendor": "0001"
        },
        "mode": "experimental_autoexposure2D",
        "processing": {
          "extrinsicHeadToUser": {
            "rotX": 0.0,
            "rotY": 0.0,
            "rotZ": 0.0,
            "transX": 0.0,
            "transY": 0.0,
            "transZ": 0.0
          },
          "version": {
            "major": 0,
            "minor": 0,
            "patch": 0
          }
        },
        "state": "RUN"
      }
    }
  }
  
  and also launch the command as roslaunch ifm3d_ros_example camera_2d.launch pcic_port:=50015 
  
  but there is no changes..

dekhanra, 2023-06-19:
Could you please mail us at support.robotics@ifm.com and we can look at the issue together.

Gabiriyal, 2023-06-19:
ok thank u

dekhanra, 2023-06-19:
The firmware installed on your VPU is 0.15.3 and the current ifm3d version on your system is 1.2.6 which is compatible with latest firmware 1.0.14. Could you also check the article number on your VPU. From the JSON config it is `M03903` which is nbo longer supported by latest firmware

dekhanra, 2023-06-19:
device

Gabiriyal, 2023-06-19:
ok

Gabiriyal, 2023-06-19:
@dekhanra  i send the mail to [support.robotics@ifm.com](mailto:support.robotics@ifm.com) .please provide me the solution for that.thanks for the replay

dekhanra, 2023-06-22:
@Gabiriyal,

As this GitHub issue has been moved to a private email thread, we are closing this issue.
If you have any outstanding questions, please contact the support email address.

Gabiriyal, 2023-06-30:
hi
thank u for the support and i read the data from all 4 cameras ,i want to merge the point cloud(i am using this version of ifm3d-ros package - https://github.com/ifm/ifm3d-ros/tree/dev1.2 

how to merge the point cloud of all 4 camera

dekhanra, 2023-06-30:
Hi @Gabiriyal,

Could you be more specific on merging the point clouds.