ifm/ifm3d-ros #54: O3R Reconnection not consistent across different streams

rpangrazio, 2022-08-08:
When the O3R VPU is power cycled, the 3d topics begin to publish automatically. The 2d rgb topics do not. They are listed in rostopic list, but no images are being published. 

To replicate bring up the system, bring up the ros nodes. Power off the VPU, and then back on. Check the topics being published. The point cloud will be published, the 2d rgb image topics will not. Do not restart the ros system.

lola-masson, 2022-08-08:
Thanks for reporting this. I think this is due to a known issues in FW 14.23 where the 2D imagers need to be switched from RUN to CONF and then back to RUN. I will double check that this is really what it is.

dekhanra, 2023-06-22:
@rpangrazio,