ifm/ifm3d-ros #56: O3D303 report an error

renguorong, 2022-10-24:
[ERROR] [1666592213.726236413]: Unsupported pixel format 3 for point cloud

renguorong, 2022-10-24:
I used the same ros launch file,O3X100 was fine ,but O3D303 reported an error

desengph, 2022-10-24:
@renguorong 
Please use the tagged release v0.7.0 for O3D and O3X.
https://github.com/ifm/ifm3d-ros/releases/tag/v0.7.0

The master branch is WIP and only supports the O3R at the moment.

renguorong, 2022-10-24:
@desengph 
Thanks, and which version is ifm3d using to match ifm3d-ros_v0.7.0?

desengph, 2022-10-24:
Version 0.7.0 is identical to release 0.6.x

To see the full documentation please check the included README and docs for the tagged release version.

Software Compatibility Matrix
=============================
<table>
  <tr>
    <th>ifm3d-ros version</th>
    <th>ifm3d version</th>
    <th>ROS distribution(s)</th>
  </tr>
  <tr>
    <td>0.1.0</td>
    <td>0.1.0</td>
    <td>Kinetic</td>
  </tr>
  <tr>
    <td>0.2.0</td>
    <td>0.2.0</td>
    <td>Kinetic, Indigo</td>
  </tr>
  <tr>
    <td>0.3.0</td>
    <td>0.2.0</td>
    <td>Kinetic, Indigo</td>
  </tr>
  <tr>
    <td>0.4.0</td>
    <td>0.2.0, 0.3.0, 0.3.1, 0.3.2</td>
    <td>Kinetic, Indigo</td>
  </tr>
  <tr>
    <td>0.4.1</td>
    <td>0.3.3, 0.4.0, 0.5.0, 0.6.0, 0.7.0, 0.8.1</td>
    <td>Kinetic, Indigo</td>
  </tr>
  <tr>
    <td>0.4.2</td>
    <td>0.9.0</td>
    <td>Kinetic</td>
  </tr>
  <tr>
    <td>0.5.0</td>
    <td>0.9.0, 0.9.1</td>
    <td>Kinetic</td>
  </tr>
  <tr>
    <td>0.5.1</td>
    <td>0.9.2</td>
    <td>Kinetic, Melodic</td>
  </tr>
  <tr>
    <td>0.6.0</td>
    <td>0.9.2, 0.9.3, 0.10.0, 0.11.0, 0.11.2</td>
    <td>Kinetic, Melodic</td>
  </tr>
  <tr>
    <td>0.6.1, 0.6.2</td>
    <td>0.11.2, 0.12.0, 0.17.0</td>
    <td>Kinetic, Melodic</td>
  </tr>
</table>

vtretyakov, 2022-11-23:
@LotfiZ I wrote the following steps for myself:
```
git clone https://github.com/ifm/ifm3d.git 
cd ifm3d
git reset --hard v0.17.0
sudo apt-get update && sudo apt-get -y upgrade
sudo apt-get update && sudo apt-get install -y git jq libssl-dev libcurl4-openssl-dev libgtest-dev libgoogle-glog-dev  libxmlrpc-c++8-dev libproj-dev build-essential coreutils cmake
sudo apt-get update && sudo apt-get install -y libopencv-dev libpcl-dev 
mkdir build
cd build
cmake -DCMAKE_INSTALL_PREFIX=/usr -DBUILD_MODULE_OPENCV=ON  -DBUILD_IN_DEPS=OFF ..
cmake --build .
sudo cmake --build . --target install

cd && cd catkin_ws/src
git clone https://github.com/ifm/ifm3d-ros.git
cd ifm3d-ros
git reset --hard v0.6.2
cd ../..
catkin_make
```
That is the latest combination that worked for me. Anything after those tags fails.

LotfiZ, 2022-11-24:
Thank you so much @vtretyakov. However when i launch the node i have this error : 
[image]

When i ping the camera i have a "Time to live exceeded". anybody has an idea what's going on here ? 

Thank you

graugans, 2022-11-25:
Hej @LotfiZ have you checked the [troubleshooting](https://github.com/ifm/ifm3d/blob/main/doc/troubleshoot.md) guide already? Could you please provide more details about your Network set-up to dig deeper into this?

LotfiZ, 2022-11-29:
Hi @graugans, 
Little bit of context : I have an IFM camera (O3D304) used to detect loads with an IFM Controller. Everything is working well from the PLC side, so in same time as we have an Industrial PC, we wanted to have data of the camera in ROS Melodic (to perform some tasks). I can ping the IP of the camera without problem but when i launch the ROS node, the message showed above pop. Is it possible to access to the data even if the camera is used for another application ? Thank you

lola-masson, 2022-11-30:
Hi @LotfiZ,
You should be able to access the camera from the IPC as well, but there are a couple things to consider. First, are you using the default IP? If not, you will need to provide the IP as a parameter when launching the ROS node. Then, is your load detection application using the O3D in "trigger" mode or in "continuous" mode? If you are using the trigger, no data is produced out except when a trigger signal is sent, which would also cause the ROS node to show a timeout error.

desengph, 2022-12-07:
Hi @LotfiZ 
please consider that the three exclusive "states" for getting data for the O3D: 
1. hardware trigger
2. software trigger
3. Continuous / asynchronous

If you have configured the camera to be in state 1/2 the data will only be available after a trigger signal. No data is streamed in the meanwhile.  
In ROS the data would only be streamed continuously if the camera is continuous mode.   
A software trigger option is available via the trigger service and is propagated to the API / system if software trigger is configured for the system.

LotfiZ, 2022-12-07:
Thank you for all this precious details, however is it possible to do the following : Configure the O3D in continous mode, then enable sw trigger in PLC side when we want to detect a specific pallet and in same time having streaming data in ROS ? in conclusion, we want to keep the PLC application of the camera as it is while accessing data flow in ROS to make other things.

graugans, 2022-12-07:
> Thank you for all this precious details, however is it possible to do the following : Configure the O3D in continous mode, then enable sw trigger in PLC side when we want to detect a specific pallet [...]

I am not sure if I got your point. You basically want to have the camera in streaming mode and capture a single frame in the PLC? It depends on your timing expectations. You can for example configure the device in continuous mode and just enable the data output in the PLC when the PLC wants data. This is a single PCIC command in the PLC. But the data will have a jitter of the frametime. If you wan more precise timing this is only possible if you do a software or hardware trigger.

LotfiZ, 2022-12-07:
Thank you Christian and all, its quite clear now !

dekhanra, 2023-06-22:
@LotfiZ @vtretyakov @renguorong 

As there have been no recent updates to this GitHub issue, we're closing it.
If you still have questions, please feel free to open a new issue thread.

LotfiZ, 2023-06-22:
Hi @dekhanra, seems legit ! Thank you for the support