ifm/ifm3d-ros #61: Error message when catkin_make. ifm3d version 0.93.0, ifm3d-ros version v0.5.1   Camera is 03X100

coa06yl, 2022-11-07:
[image]

desengph, 2022-11-07:
@coa06yl 
The ifm3d-ros version 0.5.1 requires [ifm3d version 0.9.2](https://github.com/ifm/ifm3d/releases/tag/v0.9.2)
Please see this link to see the [compatibility matrix for version 0.5.1](https://github.com/ifm/ifm3d-ros/tree/v0.6.2#software-compatibility-matrix)

Hope this helps. If you have any additional questions, please let us know.

coa06yl, 2022-11-07:
@desengph Hi Philipp, many thanks for the information. I manage to get data from RVIZ, but I keep receiving the message below. 
[image]

desengph, 2022-11-08:
@coa06yl 
This seems to me that you can not actually connect to the data stream. 
The ROS node and ROS environment seems to work fine, but the node restarts after not receiving any data.  

Can you please double-check the ROS node parameters: if you don't use the default values, they might be required to be updated: https://github.com/ifm/ifm3d-ros/tree/v0.6.0#parameters

coa06yl, 2022-11-10:
@desengph  Hi Philipp, previously, the only thing I changed was the camera IP address. Today, I reset the O3X100 camera to factory setting.  I still got the same warning msg. but I can get live image in RVIZ.  Thanks for your help!

dekhanra, 2023-06-22:
@coa06yl 
As there have been no recent updates to this GitHub issue, we're closing it.
If you still have questions, please feel free to open a new issue thread.