ifm/ifm3d-ros #62: ifm3d could not find component: stlimage

vtretyakov, 2022-11-07:
I tried compiling and installing the latest state and the latest release tag of ifm3d, including Debian packages attached to the release. No luck so far. Where is the 'stlimage' component coming from? How to get it installed or compiled?

desengph, 2022-11-08:
Hi @vtretyakov 
the cmake flags (also the stlimage flag) are documented here: http://ifm3d.com/sphinx-doc/build/html/ifm3d/doc/sphinx/content/installation_instructions/source_build.html#overview-of-the-available-build-flags
Is this what you're missing?

The Debian package (cmake repackage step) might not work. Please double check our Dockerfiles on ifm3d for general instructions.
https://github.com/ifm/ifm3d/blob/main/docker/ifm-robotics/docker/Dockerfile

lola-masson, 2022-11-09:
@vtretyakov  note that stlimage is a component of ifm3d 0.93.0, which is the version compatible with ifm3d-ros. If you installed ifm3d from main, you will have ifm3d 1.0.1, which is not yet compatible with ifm3d-ros and does not include stlimage.

dekhanra, 2023-06-22:
@vtretyakov ,
As there have been no recent updates to this GitHub issue, we're closing it.
If you still have questions, please feel free to open a new issue thread.