ifm/ifm3d-ros #63: Timeout waiting for camera!

vtretyakov, 2022-11-08:
I've been trying for 2 days to bring up the OVP800 unit with a O3R225 camera with no success. I tried following all possible confusing and contradicting instructions, including connecting cameras as described in the Hardware Unboxing, namely 2D cable to Port0, 3D cable to Port2, as well as pairwise as described in other guideline, namely 2D cable to Port0 and 3D cable to Port1. 

On the software side I used your compatibility matrix and compiled:
ifm3d v0.91.0 + ifm3d-ros v1.0.0
ifm3d v0.93.0 + ifm3d-ros v1.0.1

The result is the same:

> [ INFO] [1667904385.796269644]: Loading nodelet /ifm3d_ros_examples/camera of type ifm3d_ros/camera_nodelet to manager camera_standalone_nodelet with the following remappings:
[ INFO] [1667904385.803766394]: waitForService: Service [/ifm3d_ros_examples/camera_standalone_nodelet/load_nodelet] has not been advertised, waiting...
[ INFO] [1667904385.828006897]: Initializing nodelet with 1 worker threads.
[ INFO] [1667904385.847170178]: waitForService: Service [/ifm3d_ros_examples/camera_standalone_nodelet/load_nodelet] is now available.
[ INFO] [1667904385.867199504]: IP default: 192.168.0.69, current 192.168.0.69
[ INFO] [1667904385.867980863]: pcic port check: current 50010, default 50010
[ INFO] [1667904385.964625342]: Running dtors...
[ INFO] [1667904385.964667762]: Initializing camera...
[ INFO] [1667904386.985657382]: Initializing framegrabber...
[ INFO] [1667904386.987635215]: Nodelet arguments: 16, 50010
[ INFO] [1667904386.988019139]: Initializing image buffer...
[ WARN] [1667904387.489142860]: Timeout waiting for camera!
[ WARN] [1667904387.989997008]: Timeout waiting for camera!
[ WARN] [1667904388.490464163]: Timeout waiting for camera!
[ WARN] [1667904388.991586166]: Timeout waiting for camera!
[ WARN] [1667904389.494529229]: Timeout waiting for camera!
[ WARN] [1667904389.995031032]: Timeout waiting for camera!
[ WARN] [1667904390.495960401]: Timeout waiting for camera!
[ WARN] [1667904390.996422148]: Timeout waiting for camera!
[ WARN] [1667904391.497560812]: Timeout waiting for camera!
[ WARN] [1667904391.998816841]: Timeout waiting for camera!
[ WARN] [1667904391.999247200]: Attempting to restart framegrabber...

Is there a clear path to bring up the unit with ROS to at least see the camera data?

desengph, 2022-11-08:
Hi @vtretyakov 
can you please double-check that the camera stream you're trying to access is actually in RUN state: http://ifm3d.com/sphinx-doc/build/html/ifm3d/doc/sphinx/content/examples/basic_lib_usage/first_steps_ifm3dpy.html#run-conf-idle

The RUN state is also mirrored in the SOFT-Start helper.

Please let me know if this changes the output. If you have any additional questions, please get in touch.

vtretyakov, 2022-11-08:
Hi @desengph 
I tried following the guides there. First, configuring the camera as described [here](http://ifm3d.com/sphinx-doc/build/html/ifm3d/doc/sphinx/content/examples/o3r/configuration/configuration.html#how-to-configure-the-camera
). The output looks like that:

>  {
    "euphrates": "1.4.3",
    "firmware": "0.14.23-493",
    "kernel": "4.9.140-l4t-r32.4+g1582a8a5405d",
    "l4t": "r32.4.3",
    "schema": "v0.2.4",
    "tcu": "1.0.2"
}
RUN
great_o3r

It feels to me like the sate is RUN as required. Trying to read the image as in the link you provided gives still an error:

> [ok, frame] = fg.wait_for_frame().wait_for(500)
importlib._bootstrap.Error: Lib: Image chunk not found

The ROS node is dying:

> Initializing framegrabber...
terminate called after throwing an instance of 'std::invalid_argument'
  what():  stoi
[ifm3d_ros_examples/camera_standalone_nodelet-2] process has died [pid 5433, exit code -6, cmd /opt/ros/noetic/lib/nodelet/nodelet manager __name:=camera_standalone_nodelet

lola-masson, 2022-11-09:
Could you please send the full json configuration of your device (the output of o3r.get() with ifm3dpy for instance, or of `ifm3d dump` from the command line interface). From the snippet that you sent is seems like the "RUN" setting is not applied properly.  You should have something like: 
```json
    "port3": {
...
      "state": "RUN"
    },
```

vtretyakov, 2022-11-13:
@lola-masson please mind that we are debugging ifm3d-ros! Are there steps to be performed using ROS? Can the ROS node switch the camera into the RUN state or perform required configurations? If not, then when will it be possible?

lola-masson, 2022-11-14:
Yes, there is a service that you can use for configuration, see [here](https://ifm3d.com/sphinx-doc/build/html/ROS/ifm3d-ros/ifm3d_ros_driver/doc/dump_and_config.html#config)

samialperen, 2023-01-13:
hi @lola-masson, I am having the same error with ifm3d v0.93.0 + ifm3d-ros v1.0.1 configuration (ROS noetic). I upgraded firmware for OVP800 too. The camera I am using is O3R225. Do you have any suggestions?

GiovanniBalestrieri, 2023-02-25:
Good evening,
I have the same problem with an O3R225 sensor and OVP800 video processing unit. I have connected the 3D cable to port0 and the 2D cable to port2. I am trying to publish the pointcloud and visualize it with RViz. I am able to reproduce the problem and I am willing to provide information as you ask them. Thank you for your support. 

When I launch the camera node I get the `timeout waiting for camera`  message (CF: end of the post). If I launch the visualization node, Rviz shows an empty pointcloud and no image is shown. 

The default configuration shows that both sensors are in CONF state as you mentioned.

```
"ports": {
    "port0": {
       "state": "CONF"
    },
    "port2": {
    "state": "CONF"
    }
}
```

When I try to set the state to "RUN", the ROS2 node fails and throws an error code. I tried three approaches to change the state and all of them led to the same error.

Attempt 1 using general purpose service to modify the configuration of the sensor:
`ros2 service call /ifm3d/camera/Config ifm3d_ros2/srv/Config "{json: '{\"ports\":{\"port0\":{\"state\":\"RUN\"}}}'}"`
Attempt 2 by changing the json values:
`ifm3d dump | jq '(.ports[] | select(.info.features.type == "3D") | .state ) = "RUN"' | ifm3d config`
Attempt 3 using dedicated softon service:
`ros2 service call /ifm3d/camera/Softon ifm3d_ros2/srv/Softon`

As soon as I execute the command, the `camera_standalone-1` node dies with this message:
`[ERROR] [camera_standalone-1]: process has died [pid 397066, exit code -11, cmd '/home/userk/development/ros2/galactic_ws/install/ifm3d_ros2/lib/ifm3d_ros2/camera_standalone --ros-args -r __node:=camera -r __ns:=/ifm3d'].`

Here is the output of `ros2 launch ifm3d_ros2 camera_managed.launch.py` command:
```
[INFO] [launch]: All log files can be found below /home/userk/.ros/log/2023-02-25-18-34-25-407446-dopamine-396506
[INFO] [launch]: Default logging verbosity is set to INFO
2023-02-25 18:34:25,563 - Publishing tf2 transform from camera_optical_link to camera_link
2023-02-25 18:34:25,563 - Publishing tf2 transform from map to camera_optical_link
[WARNING] [launch_ros.actions.node]: Parameter file path is not a file: .
[INFO] [camera_standalone-1]: process details: cmd=[/home/userk/development/ros2/galactic_ws/install/ifm3d_ros2/lib/ifm3d_ros2/camera_standalone, --ros-args, -r, __node:=camera, -r, __ns:=/ifm3d], cwd='None', custom_env?=False
[INFO] [ros2-2]: process details: cmd=[ros2, run, tf2_ros, static_transform_publisher, 0, 0, 0, 0, 0, 0, camera_optical_link, camera_link], cwd='None', custom_env?=False
[INFO] [ros2-3]: process details: cmd=[ros2, run, tf2_ros, static_transform_publisher, 0, 0, 0, 0, 0, 0, map, camera_optical_link], cwd='None', custom_env?=False
[INFO] [camera_standalone-1]: process started with pid [396517]
[INFO] [ros2-2]: process started with pid [396519]
[INFO] [ros2-3]: process started with pid [396521]
[camera_standalone-1] [INFO] 1677346465.652949691 [ifm3d.camera]: namespace: /ifm3d
[camera_standalone-1] [INFO] 1677346465.653824268 [ifm3d.camera]: node name: camera
[camera_standalone-1] [INFO] 1677346465.653862403 [ifm3d.camera]: middleware: rmw_cyclonedds_cpp
[camera_standalone-1] [INFO] 1677346465.653885779 [ifm3d.camera]: camera frame: camera_link
[camera_standalone-1] [INFO] 1677346465.653903954 [ifm3d.camera]: optical frame: camera_optical_link
[camera_standalone-1] [INFO] 1677346465.653927431 [ifm3d.camera]: declaring parameters...
[camera_standalone-1] [INFO] 1677346465.654666424 [ifm3d.camera]: After the parameters declaration
[camera_standalone-1] [INFO] 1677346465.659860526 [ifm3d.camera]: After publishers declaration
[camera_standalone-1] [INFO] 1677346465.661669841 [ifm3d.camera]: node created, waiting for `configure()`...
[camera_standalone-1] [INFO] 1677346465.879770101 [ifm3d.camera]: on_configure(): unconfigured -> configuring
[camera_standalone-1] [INFO] 1677346465.879818153 [ifm3d.camera]: Parsing parameters...
[camera_standalone-1] [INFO] 1677346465.879877321 [ifm3d.camera]: pcic_port: 50010
[camera_standalone-1] [INFO] 1677346465.879912490 [ifm3d.camera]: ip: 192.168.0.69
[camera_standalone-1] [INFO] 1677346465.879930444 [ifm3d.camera]: xmlrpc_port: 80
[camera_standalone-1] [INFO] 1677346465.879951737 [ifm3d.camera]: password: 
[camera_standalone-1] [INFO] 1677346465.879967490 [ifm3d.camera]: schema_mask: 15
[camera_standalone-1] [INFO] 1677346465.879986137 [ifm3d.camera]: timeout_millis: 500
[camera_standalone-1] [INFO] 1677346465.880007847 [ifm3d.camera]: timeout_tolerance_secs: 5.000000
[camera_standalone-1] [INFO] 1677346465.880028302 [ifm3d.camera]: frame_latency_thresh (seconds): 1.000000
[camera_standalone-1] [INFO] 1677346465.880049174 [ifm3d.camera]: sync_clocks: false
[camera_standalone-1] [INFO] 1677346465.880063140 [ifm3d.camera]: Parameters parsed OK.
[camera_standalone-1] [INFO] 1677346465.880075003 [ifm3d.camera]: Initializing camera...
[camera_standalone-1] [INFO] 1677346465.897108892 [ifm3d.camera]: Initializing FrameGrabber with mask: 15
[camera_standalone-1] [INFO] 1677346465.897750788 [ifm3d.camera]: Initializing ImageBuffer...
[INFO] [launch.user]: Emitting 'TRANSITION_ACTIVATE' event
[camera_standalone-1] [INFO] 1677346465.897879068 [ifm3d.camera]: Configuration complete.
[camera_standalone-1] [INFO] 1677346465.901145599 [ifm3d.camera]: on_activate(): inactive -> activating
[... 26 lines cut ...]
[INFO] [launch.user]: Emitting 'TRANSITION_CONFIGURE' event
[... cut ...]
[camera_standalone-1] [INFO] 1677346470.938199160 [ifm3d.camera]: on_activate(): inactive -> activating
[camera_standalone-1] [INFO] 1677346470.938242813 [ifm3d.camera]: Activating publishers...
[camera_standalone-1] [INFO] 1677346470.938259816 [ifm3d.camera]: Publishers activated.
[camera_standalone-1] [INFO] 1677346470.938364799 [ifm3d.camera]: Starting publishing loop...
[camera_standalone-1] [WARN] 1677346471.438601288 [ifm3d.camera]: Timeout waiting for camera!
[camera_standalone-1] [WARN] 1677346471.938949592 [ifm3d.camera]: Timeout waiting for camera!
```

GiovanniBalestrieri, 2023-02-25:
hello @vtretyakov the link provided by @lola-masson is not available anymore. I used a dedicated service to change the state from `CONF` to `RUN`.

I used the following service call. 
`ros2 service call /ifm3d/camera/Config ifm3d_ros2/srv/Config "{json: '{\"ports\":{\"port0\":{\"state\":\"RUN\"}}}'}"`

Unfortunately, this command is causing the camera node to fail but it should do the trick.

lola-masson, 2023-03-01:
Hi all,

Unfortunately the ROS notes are currently behind in terms of version compatibility, compared to our other packages, which explains some of these headaches especially with newer firmwares. Sorry for the inconvenience.

A good way to verify that the issue is on the side of ROS and not of the O3R is to use the ifm Vision Assistant to check the data streams (you can also use python or c++, but the Vision Assistant provides a convenient GUI). If you all can email support.robotics@ifm.com requesting the Vision Assistant we can send you a link to the latest version. Alternatively, you can go to ifm.com, find the O3D303, go to the download section and download version 2.4.13 from there (it is an older version but should have basic functionalities for the O3R). Once we have identified which side the issues comes from, it will be easier to provide guidance.

dekhanra, 2023-06-22:
@vtretyakov 
As there have been no recent updates to this GitHub issue, we're closing it.
If you still have questions, please feel free to open a new issue thread.