ifm/ifm3d-ros #65: IFM 1.1.1 / Ubuntu 22.04 support

maxlein, 2022-12-12:
When can we expect an update to support latest ifm3d release 1.1.1?   
As ros2 galactic has reached end-of-life we would need this pretty soon.

lola-masson, 2022-12-12:
Hey @maxlein, we are planning to work on this in January. Sorry for the inconvenience.

maxlein, 2023-02-09:
Any info on this?
Do you have a more specific time frame to work with?

lola-masson, 2023-02-09:
Hi @maxlein, unfortunately we had to delay working on this until mid-March. Let us know if we can find a workaround for you in the meantime.

dekhanra, 2023-06-22:
@maxlein,

As there have been no recent updates to this GitHub issue, we're closing it.
If you still have questions, please feel free to open a new issue thread.