ifm/ifm3d-ros #66: Compiling everything and making O3Rxx Camera work With ROS Noetic

samialperen, 2023-01-12:
Heyo, 

After spending countless hours, I still  either could not compile ifm3d and ros wrapper or I compiled and I can not connect the camera. 

Here are some of the stuff I tried: 
* https://github.com/ifm/ifm3d/issues/380
*  Trying ifm3d different tags --> always same error as mentioned in above issue
*  Installed ros-noetic-ifm3d with apt --> when trying to connect camera getting this error ```[ INFO] [1673553778.279907362]: Loading nodelet /ifm3d_ros_examples/camera of type ifm3d_ros/camera_nodelet to manager camera_standalone_nodelet with the following remappings:
[ INFO] [1673553778.282253150]: waitForService: Service [/ifm3d_ros_examples/camera_standalone_nodelet/load_nodelet] has not been advertised, waiting...
[ INFO] [1673553778.292271043]: Initializing nodelet with 16 worker threads.
[ INFO] [1673553778.303088068]: waitForService: Service [/ifm3d_ros_examples/camera_standalone_nodelet/load_nodelet] is now available.
[ INFO] [1673553778.357031416]: onInit(): /ifm3d_ros_examples/camera
[ INFO] [1673553778.405622412]: Camera clock will not be sync'd to system clock
[ INFO] [1673553778.405655652]: Running dtors...
[ INFO] [1673553778.405669190]: Initializing camera...
[ INFO] [1673553779.411659474]: Initializing framegrabber...
terminate called after throwing an instance of 'std::invalid_argument'
  what():  stoi
[ifm3d_ros_example```

Can someone help me out? Many of the documentation on the GitHub is either deprecated or does not exist. Some of the links mentioned in previous issues does not exist anymore. 

Thank you very much for your help!

lola-masson, 2023-01-13:
@samialperen let's get your through the ifm3d build first, in issue https://github.com/ifm/ifm3d/issues/380. Once you have that ready to go, we can move on to ifm3d-ros installation. Please note that the ros-noetic-ifm3d package is unfortunately deprecated.

ifm-csr, 2023-01-16:
Hi @samialperen. It still could be that the camera is in "CONF" mode. You should switch each port to "RUN". See the documentation about the Dump and Config services here: https://ifm3d.com/sphinx-doc/build/html/ROS/ifm3d-ros/ifm3d_ros_driver/doc/dump_and_config.html. In doubt, send us the output of the Dump service, so that we can look at the current configuration of the device.

samialperen, 2023-01-16:
@ifm-csr  @lola-masson 
I changed port0 and port2 state to RUN and got this error now:
```
[ WARN] [1673895056.337101039]: -100010: Lib: Pixel format error - didn't expect a particular pixel type
[ WARN] [1673895056.337145282]: Timeout waiting for camera!
```

You can see dump.json attached as text file
[dump-copy.txt](https://github.com/ifm/ifm3d-ros/files/10428638/dump-copy.txt)

samialperen, 2023-01-17:
Hi @lola-masson , it works for me. Thanks a lot.

dekhanra, 2023-06-22:
@samialperen,

As there have been no recent updates to this GitHub issue, we're closing it.
If you still have questions, please feel free to open a new issue thread.