ifm/ifm3d-ros #68: Lib: Pixel format error

thermofluid06, 2023-03-01:
Hi everyone, 

 I am using ifm3d version 0.93.0, ifm3d-ros version v1.0.1, camera is 03R222

I have the following warning message when I try to launch the camera in ROS and I can't see any topic being published in ROS. Rviz is also empty. Can you help me solving the issue?:

`[ WARN] [1677676864.055349177]: Timeout waiting for camera!
[ WARN] [1677676864.105023235]: -100010: Lib: Pixel format error - didn't expect a particular pixel type
[ WARN] [1677676864.105140821]: Timeout waiting for camera!
[ WARN] [1677676864.155067663]: -100010: Lib: Pixel format error - didn't expect a particular pixel type
`

[image]

lola-masson, 2023-03-01:
Which firmware version are you using?

Also, please check that the camera connected to Port0 (the one you are requesting data from in the screenshot) is in "RUN" mode. If the camera in Port0 is the RGB camera, and you are using FW 16.23, then this is a known bug, see https://github.com/ifm/ifm3d-ros/issues/67.

thermofluid06, 2023-03-01:
I can ping the device but if I use ifm3d without the ROS wrapper, I have the following response from the camera:

[image]

thermofluid06, 2023-03-01:
> Which firmware version are you using?
> 
> Also, please check that the camera connected to Port0 (the one you are requesting data from in the screenshot) is in "RUN" mode. If the camera in Port0 is the RGB camera, and you are using FW 16.23, then this is a known bug, see #67.

I can confirm that we are using FW 16.23 after running `ifm3d dump`.

lola-masson, 2023-03-01:
> I can ping the device but if I use ifm3d without the ROS wrapper, I have the following response from the camera:
> 
> [image]

Since ifm3d supports multiple devices, it happens that some available commands are not applicable to the O3R, which is what you are seeing here.

> I can confirm that we are using FW 16.23 after running `ifm3d dump`.

Could you please provide the output of the ifm3d dump here, thanks!

thermofluid06, 2023-03-02:
Here is the output from the ifm3d dump. Sorry for the formatting.

`{
  "applications": {
    "classes": {
      "ods": {
        "instanceSchema": {},
        "license": "UNLICENSED",
        "minimumFirmwareVersion": {
          "major": 0,
          "minor": 16,
          "patch": 3
        },
        "version": {
          "major": 0,
          "minor": 0,
          "patch": 0
        }
      }
    }
  },
  "device": {
    "clock": {
      "currentTime": 1675546327159571032,
      "sntp": {
        "active": true,
        "activeServerAddress": "",
        "activeServerName": "time1.google.com",
        "availableServers": [],
        "systemClockSynchronized": false
      }
    },
    "diagnostic": {
      "confInitStages": [
        "device",
        "ports",
        "applications"
      ],
      "temperatures": [
        {
          "entity": "PLL-therm",
          "value": 42.5
        },
        {
          "entity": "MCPU-therm",
          "value": 42.5
        },
        {
          "entity": "PMIC-Die",
          "value": 100.0
        },
        {
          "entity": "Tboard_tegra",
          "value": 38.0
        },
        {
          "entity": "GPU-therm",
          "value": 40.5
        },
        {
          "entity": "BCPU-therm",
          "value": 42.5
        },
        {
          "entity": "thermal-fan-est",
          "value": 41.70000076293945
        },
        {
          "entity": "Tdiode_tegra",
          "value": 39.75
[... 227 lines cut ...]
          "minAmplitude": 20.0,
          "minReflectivity": 2.0,
          "mixedPixelFilterMode": 1,
          "mixedPixelThresholdRad": 0.05
        },
        "extrinsicHeadToUser": {
          "rotX": 0.0,
          "rotY": 0.0,
          "rotZ": 0.0,
          "transX": 0.0,
          "transY": 0.0,
          "transZ": 0.0
        },
        "version": {
          "major": 0,
          "minor": 13,
          "patch": 12
        }
      },
      "state": "RUN"
    }
  }
}

`

lola-masson, 2023-03-02:
When launching the ROS node with the camera.launch, you are pointing to port0 by default. Which in your case is a 2D camera, hence the timeout (known bug https://github.com/ifm/ifm3d-ros/issues/67). To run the camera.launch pointing to port 2, you can use: 
`roslaunch ifm3d_ros_driver camera.launch pcic_port:=50012`

thermofluid06, 2023-03-07:
I have run the launch file with the port argument (pcic_port:=50012) as you have suggested. This time I got the following warning message. In Rviz `rostopic echo /ifm3d_ros_driver/camera/cloud ` doesn't return any measurement data. I have tested with two different computers.
[image]

lola-masson, 2023-03-09:
This times it looks like you are pointing to the right port indeed. Have you been able to get data without using ROS, for instance using ifm3dpy, or the ifm Vision Assistant?

thermofluid06, 2023-03-20:
I tried again `roslaunch ifm3d_ros_driver camera.launch pcic_port:=50012` and I got the following response below. However I can ping the camera. Later I tried to change the json file so that the port0 switches to RUN state using `rosservice call /ifm3d_ros_driver/camera/Config "json: 'test.json'"` . When I call the service, the process which is launching the camera dies. 

I used the instructions from this link to install C++API : [https://api.ifm3d.com/html/content/installation_instructions/install_from_source_linux.html](https://api.ifm3d.com/html/content/installation_instructions/install_from_source_linux.html) 

[image]
[image]

lola-masson, 2023-03-22:
In the latest screenshot, it looks like you cannot ping the camera. Did you happen to change the IP address of the device? And again, are you able to get data in any other way than ROS (python, Vision Assistant or c++)?