ifm/ifm3d-ros #69: Connection issue with the sensor

himanshuak, 2023-03-28:
Camera: M03969
lib version : ifm3d_0.93.0
ros: noetic
ifm ros: v1.0.1

Hello, 
I am trying use the ifm M03969 sensor for an academic project which requires to run the camera using ros. I am facing connection issue as shown in the below snippet
````
$ roslaunch ifm3d_ros_examples camera.launch 
... logging to /home/himanshu/.ros/log/f4212924-cd62-11ed-9ea7-5f85a4eb72de/roslaunch-himanshu-HP-Pavilion-Gaming-Laptop-15-dk0xxx-110123.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://himanshu-HP-Pavilion-Gaming-Laptop-15-dk0xxx:43231/

SUMMARY
========

PARAMETERS
 * /ifm3d_ros_examples/camera/amplitude/disable_pub_plugins: ['image_transport...
 * /ifm3d_ros_examples/camera/assume_sw_triggered: False
 * /ifm3d_ros_examples/camera/confidence/disable_pub_plugins: ['image_transport...
 * /ifm3d_ros_examples/camera/distance/disable_pub_plugins: ['image_transport...
 * /ifm3d_ros_examples/camera/frame_id_base: ifm3d_ros_example...
 * /ifm3d_ros_examples/camera/frame_latency_thresh: 60.0
 * /ifm3d_ros_examples/camera/good_bad_pixels/disable_pub_plugins: ['image_transport...
 * /ifm3d_ros_examples/camera/ip: 192.168.0.69
 * /ifm3d_ros_examples/camera/password: 
 * /ifm3d_ros_examples/camera/pcic_port: 50010
 * /ifm3d_ros_examples/camera/raw_amplitude/disable_pub_plugins: ['image_transport...
 * /ifm3d_ros_examples/camera/schema_mask: 15
 * /ifm3d_ros_examples/camera/soft_off_timeout_millis: 500
 * /ifm3d_ros_examples/camera/soft_off_timeout_tolerance_secs: 600.0
 * /ifm3d_ros_examples/camera/soft_on_timeout_millis: 500
 * /ifm3d_ros_examples/camera/soft_on_timeout_tolerance_secs: 5.0
 * /ifm3d_ros_examples/camera/timeout_millis: 500
 * /ifm3d_ros_examples/camera/timeout_tolerance_secs: 5.0
 * /ifm3d_ros_examples/camera/xmlrpc_port: 80
 * /ifm3d_ros_examples/camera/xyz_image/disable_pub_plugins: ['image_transport...
 * /rosdistro: noetic
 * /rosversion: 1.16.0

NODES
  /ifm3d_ros_examples/
    camera (nodelet/nodelet)
    camera_standalone_nodelet (nodelet/nodelet)
    camera_tf (tf2_ros/static_transform_publisher)

auto-starting new master
process[master]: started with pid [110131]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to f4212924-cd62-11ed-9ea7-5f85a4eb72de
process[rosout-1]: started with pid [110141]
started core service [/rosout]
process[ifm3d_ros_examples/camera_standalone_nodelet-2]: started with pid [110148]
process[ifm3d_ros_examples/camera-3]: started with pid [110149]
process[ifm3d_ros_examples/camera_tf-4]: started with pid [110150]
[ INFO] [1680006040.974075246]: Loading nodelet /ifm3d_ros_examples/camera of type ifm3d_ros/camera_nodelet to manager camera_standalone_nodelet with the following remappings:
[ INFO] [1680006040.976826924]: waitForService: Service [/ifm3d_ros_examples/camera_standalone_nodelet/load_nodelet] has not been advertised, waiting...
[ INFO] [1680006040.985202234]: Initializing nodelet with 8 worker threads.
[ INFO] [1680006040.999133622]: waitForService: Service [/ifm3d_ros_examples/camera_standalone_nodelet/load_nodelet] is now available.
[ INFO] [1680006041.035557128]: IP default: 192.168.0.69, current 192.168.0.69
[ INFO] [1680006041.037220851]: pcic port check: current 50010, default 50010
[ INFO] [1680006041.146103593]: Running dtors...
[ INFO] [1680006041.146147135]: Initializing camera...
[ INFO] [1680006045.152094168]: Initializing framegrabber...
[ WARN] [1680006048.155031639]: -100001: Lib: XMLRPC Timeout - can you `ping' the sensor?
[ WARN] [1680006048.155358699]: Could not initialize pixel stream!
[ INFO] [1680006049.155726667]: Running dtors...
[ INFO] [1680006049.155864147]: Initializing camera...
[ INFO] [1680006053.158873536]: Initializing framegrabber...
[ WARN] [1680006056.162753930]: -100001: Lib: XMLRPC Timeout - can you `ping' the sensor?
[ WARN] [1680006056.163029729]: Could not initialize pixel stream!
[ INFO] [1680006057.163442200]: Running dtors...
[ INFO] [1680006057.163588723]: Initializing camera...
^C[ifm3d_ros_examples/camera_tf-4] killing on exit
[ifm3d_ros_examples/camera-3] killing on exit
[ifm3d_ros_examples/camera_standalone_nodelet-2] killing on exit
[ INFO] [1680006058.548733938]: Unloading nodelet /ifm3d_ros_examples/camera from manager camera_standalone_nodelet
^C[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

````
And without ros
````
$ ifm3d hz
ifm3d error: -100001
Lib: XMLRPC Timeout - can you `ping' the sensor?

````
I havent been able to find out whats causing the issue. Any suggestions would be very helpful. Thanks in advance!!

desengph, 2023-03-28:
Hi @himanshuak,
can you please double-check your network settings?
It seems you can not reach the O3R system on its default static IP: `192.168.0.69`.

If you have changed the IP settings, please include those in the respective ROS call / launch files and ifm3d CLI calls. 

Please keep in mind that the O3R system does not come up in RUN state, but in CONF state.
The user has to manually change the states to receive any data.

For further details, please see our O3R documentation [here](https://ifm3d.com/)

himanshuak, 2023-04-03:
Hi @desengph ,
Thanks for the response. I followed you suggestion and used the ifm vision software. I was able to connect to the sensor and change the ports to RUN state. I was able to visualize the sensor data. This just worked once.  I disconnected the sensor and tried to connect to it again, it didnt work.

I have also changed my pc ip address to 192.168.0.2 and yet i am not able to ping the sensor both on windows and ubuntu. Should i reinstall the firmware?

himanshuak, 2023-04-06:
I was able resolve the issue with respect to the connection. The issue was that the DHCP was enabled in the VPU and i was not able to get the right IP until I used the `arp` command. Now the DHCP is disabled and I am able to connect to the vpu. The ping works fine and I am also able to get the sensor data using the Ifm Vision Assistant software. 
When i launch the rosnode, it comes up without any errors or warnings, but  I am not able to get the sensor data. When I do rostopic echo, its an empty message
````
$roslaunch ifm3d_ros_examples camera.launch 
... logging to /home/himanshu/.ros/log/ca783506-d2f6-11ed-8895-dfba26d08b26/roslaunch-himanshu-HP-Pavilion-Gaming-Laptop-15-dk0xxx-4508.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://himanshu-HP-Pavilion-Gaming-Laptop-15-dk0xxx:34199/

SUMMARY
========

PARAMETERS
 * /ifm3d_ros_examples/camera/amplitude/disable_pub_plugins: ['image_transport...
 * /ifm3d_ros_examples/camera/assume_sw_triggered: False
 * /ifm3d_ros_examples/camera/confidence/disable_pub_plugins: ['image_transport...
 * /ifm3d_ros_examples/camera/distance/disable_pub_plugins: ['image_transport...
 * /ifm3d_ros_examples/camera/frame_id_base: ifm3d_ros_example...
 * /ifm3d_ros_examples/camera/frame_latency_thresh: 60.0
 * /ifm3d_ros_examples/camera/good_bad_pixels/disable_pub_plugins: ['image_transport...
 * /ifm3d_ros_examples/camera/ip: 192.168.0.69
 * /ifm3d_ros_examples/camera/password: 
 * /ifm3d_ros_examples/camera/pcic_port: 50010
 * /ifm3d_ros_examples/camera/raw_amplitude/disable_pub_plugins: ['image_transport...
 * /ifm3d_ros_examples/camera/schema_mask: 15
 * /ifm3d_ros_examples/camera/soft_off_timeout_millis: 500
 * /ifm3d_ros_examples/camera/soft_off_timeout_tolerance_secs: 600.0
 * /ifm3d_ros_examples/camera/soft_on_timeout_millis: 500
 * /ifm3d_ros_examples/camera/soft_on_timeout_tolerance_secs: 5.0
 * /ifm3d_ros_examples/camera/timeout_millis: 500
 * /ifm3d_ros_examples/camera/timeout_tolerance_secs: 5.0
 * /ifm3d_ros_examples/camera/xmlrpc_port: 80
 * /ifm3d_ros_examples/camera/xyz_image/disable_pub_plugins: ['image_transport...
 * /rosdistro: noetic
 * /rosversion: 1.16.0

NODES
  /ifm3d_ros_examples/
    camera (nodelet/nodelet)
    camera_standalone_nodelet (nodelet/nodelet)
    camera_tf (tf2_ros/static_transform_publisher)

auto-starting new master
process[master]: started with pid [4516]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to ca783506-d2f6-11ed-8895-dfba26d08b26
process[rosout-1]: started with pid [4526]
started core service [/rosout]
process[ifm3d_ros_examples/camera_standalone_nodelet-2]: started with pid [4533]
process[ifm3d_ros_examples/camera-3]: started with pid [4534]
process[ifm3d_ros_examples/camera_tf-4]: started with pid [4535]
[ INFO] [1680619292.396114509]: Loading nodelet /ifm3d_ros_examples/camera of type ifm3d_ros/camera_nodelet to manager camera_standalone_nodelet with the following remappings:
[ INFO] [1680619292.399191536]: waitForService: Service [/ifm3d_ros_examples/camera_standalone_nodelet/load_nodelet] has not been advertised, waiting...
[ INFO] [1680619292.403869121]: Initializing nodelet with 8 worker threads.
[ INFO] [1680619292.421940439]: waitForService: Service [/ifm3d_ros_examples/camera_standalone_nodelet/load_nodelet] is now available.
[ INFO] [1680619292.462499303]: IP default: 192.168.0.69, current 192.168.0.69
[ INFO] [1680619292.464062990]: pcic port check: current 50010, default 50010
[ INFO] [1680619292.578336712]: Running dtors...
[ INFO] [1680619292.578383133]: Initializing camera...
[ INFO] [1680619293.590663493]: Initializing framegrabber...
[ INFO] [1680619293.592028001]: Nodelet arguments: 16, 50010
[ INFO] [1680619293.592170234]: Initializing image buffer...
[ INFO] [1680619293.624670897]: Camera's time and client's time are not synced
[ INFO] [1680619293.625018193]: uvec image size: 0
[ INFO] [1680619293.625673535]: Got unit vectors, restarting framegrabber with mask: 15
[ INFO] [1680619293.625795273]: Running dtors...
[ INFO] [1680619293.626971641]: Initializing camera...
[... 179 lines cut ...]
          "medianSizeDiv2": 0,
          "minAmplitude": 20.0,
          "minReflectivity": 0.0,
          "mixedPixelFilterMode": 1,
          "mixedPixelThresholdRad": 0.15
        },
        "extrinsicHeadToUser": {
          "rotX": 0.0,
          "rotY": 0.0,
          "rotZ": 0.0,
          "transX": 0.0,
          "transY": 0.0,
          "transZ": 0.0
        },
        "version": {
          "major": 0,
          "minor": 0,
          "patch": 0
        }
      },
      "state": "RUN"
    }
  }
}
````

desengph, 2023-04-06:
Hi @himanshuak 
please be aware that all software packages are disgned to use the default static IP address of ETH0 when no different arguments are provided: for further details see this doc: https://ifm3d.com/documentation/DeviceConfiguration/Interfaces/ethernet.html?

If your device was set to DHCP this was changed by the user at some point, hence the default static IP addresses not working anymore.

In regard to your second point:
Please check the arguments of the launch file. This states that the PCIC TCP port used is 50010. This corresponds to the physical port 0 as marked on the VPU. 
Connected to port 0 is a 2D camera (stream). For 2D cameras the 3D data, e.e. cloud is not available, and the data can not be supplied via the corresponding messages. 

Please change the arguments of the node before the launch to `pcic_port:=50012` to use the camera stream connected to port2.

dekhanra, 2023-06-22:
@himanshuak

As there have been no recent updates to this GitHub issue, we're closing it.
If you still have questions, please feel free to open a new issue thread.