ifm/ifm3d-ros #70: O3X100  roslaunch ifm3d camera.launch    [WARN] : Timeout waiting for camera!

coa06yl, 2023-04-25:
[image]
[image]
[image]
[image]
[image]

The ROS node parameters are default values.  
Many thanks!

coa06yl, 2023-04-25:
ifm3d-ros version 0.5.1 and ifm3d version 0.9.2

coa06yl, 2023-04-28:
IFM_Software: 1.30.5309.  I believe this is the firmware version. This firmware version is compatible with ifm3d-ros version 0.5.1 and ifm3d version 0.9.2?    Thanks!

lola-masson, 2023-04-28:
Ok, thank you for the details. Let me get back to you as I need to reproduce with this specific version or find someone who has. Just to confirm, you can "ping" the device on 192.168.0.69?

coa06yl, 2023-04-28:
Yes. I am using the default IP address.

dekhanra, 2023-06-15:
@coa06yl, with respective to the O3X1xx cameras, the number of PCIC connections are limited to one. So, if the ifmVisionAssistant is running parallel then you may encounter this issue. So, please make sure that you are not running ifmVisionAssistant while launching the ROS nodes.

Gabiriyal, 2023-06-19:
i am using O3R camera with the ifm3d-1.2.6 and ifm3d-ros 1.2.6(pre-release version) it having errro as like this 
[ INFO] [1686975141.666864385]: Initializing camera...
[ INFO] [1686975146.677230126]: Initializing framegrabber...
[ INFO] [1686975151.677437140]: Start streaming frames
[ WARN] [1686975156.678626410]: Timeout waiting for camera!
[ WARN] [1686975156.678666220]: Attempting to restart framegrabber...
[ INFO] [1686975156.678681432]: Running dtors...
[ INFO] [1686975156.678854796]: Initializing camera...
[ INFO] [1686975161.691967945]: Initializing framegrabber...
[ INFO] [1686975166.692378127]: Start streaming frames
[ WARN] [1686975171.693039815]: Timeout waiting for camera!
[ WARN] [1686975171.693129855]: Attempting to restart framegrabber...
[ INFO] [1686975171.693171770]: Running dtors...
[ INFO] [1686975171.693577726]: Initializing camera...
[ INFO] [1686975176.705947934]: Initializing framegrabber...
[ INFO] [1686975181.706168308]: Start streaming frames
[ WARN] [1686975186.707217287]: Timeout waiting for camera!
[ WARN] [1686975186.707316279]: Attempting to restart framegrabber...
[ INFO] [1686975186.707379356]: Running dtors...
[ INFO] [1686975186.707818487]: Initializing camera...
[ INFO] [1686975191.722609885]: Initializing framegrabber...

what can i do ..please any one can replay for this solution

graugans, 2023-06-22:
Hej @Gabiriyal please open a separate issue. This is related to O3X1xx and not O3R those are completely different products and require different version.