ifm/ifm3d-ros #74: ifm3d ver 0.20.3 and ifm3d_ros ver 0.7.0, catkin_make error

coa06yl, 2023-06-01:
[image]
[image]
But when  I check the ifm3d version, it is 0.18.0.

lola-masson, 2023-06-01:
Maybe you installed ifm3d 0.18.0 from source and you still have it installed on your machine and this is the one ifm3d-ros finds.

coa06yl, 2023-06-07:
> Maybe you installed ifm3d 0.18.0 from source and you still have it installed on your machine and this is the one ifm3d-ros finds.

[image]
I follow the readme.md instruction in ver 0.20.0. after install ifm3d, the version is  0.18.0. Pls help!

coa06yl, 2023-06-08:
@dekhanra  this is where I found ifm3d ver 0.20.3
https://github.com/ifm/ifm3d/releases

[image]

dekhanra, 2023-06-08:
Hi @coa06yl ,

Sorry my bad, the arguments are initialized again as empty arguments in the Dockerfile at Line 67 and Line 43. Can you delete them and try again. Below is the Dockerfile

```docker
FROM ros:melodic-perception AS build

ARG ROS_DISTRO=melodic
ARG LSB_RELEASE=bionic
ARG CMAKE_VERSION=3.20.6
ARG IFM3D_TAG=tags/v0.20.3
ARG IFM3D_ROS_BRANCH="v0.7.0"
ARG IFM3D_ROS_CLONE_REPO=https://github.com/ifm/ifm3d-ros
ARG IFM3D_CLONE_REPO=https://github.com/ifm/ifm3d

# Create the ifm user
RUN id ifm 2>/dev/null || useradd --uid 30000 --create-home -s /bin/bash -U ifm
WORKDIR /home/ifm

ARG DEBIAN_FRONTEND=noninteractive
RUN apt-get update && \
    DEBIAN_FRONTEND=noninteractive apt-get install -y \
    build-essential \
    cmake \
    coreutils \
    findutils \
    git \
    libboost-all-dev \
    libcurl4-openssl-dev \
    libgoogle-glog-dev \
    libgtest-dev \
    libopencv-dev \
    libpcl-dev \
    libproj-dev \
    libxmlrpc-c++8-dev \
    locales \
    ninja-build \
    wget \
    gpg
RUN apt-get clean

# Install cmake
RUN wget -O - "https://github.com/Kitware/CMake/releases/download/v${CMAKE_VERSION}/cmake-${CMAKE_VERSION}-linux-$(uname -i).tar.gz" \
    | tar -xz --strip-components=1 -C /usr

# First clone ifm3d repo
RUN mkdir src && \
    cd src && \
    git clone ${IFM3D_CLONE_REPO} ifm3d && \
    cd ifm3d && \
    git checkout ${IFM3D_TAG} && \
    echo "Building from tag ${IFM3D_TAG}" && \
    mkdir build && \
    cd build && \
    cmake -GNinja cmake \
    -DCMAKE_INSTALL_PREFIX=/install \
    -DBUILD_MODULE_IMAGE=ON \
    -DBUILD_IN_DEPS=OFF .. && \
    cmake --build . && \
    cmake --build . --target install

# copy ifm from install location to /usr
RUN cp -r /install/* /usr/

# use the first build stage to also compile the ros package as it has shared dependencies as ifm3d
RUN mkdir -p catkin_ws/ifm3d-ros/src

# Initialize catkin workspace
RUN /bin/bash -c 'cd catkin_ws/ifm3d-ros/src; . /opt/ros/${ROS_DISTRO}/setup.bash; catkin_init_workspace'

# Clone and build ifm3d-ros repo
RUN cd /home/ifm/catkin_ws/ifm3d-ros/src && \
    git clone ${IFM3D_ROS_CLONE_REPO} -b ${IFM3D_ROS_BRANCH} --single-branch 

RUN /bin/bash -c 'cd catkin_ws/ifm3d-ros; . /opt/ros/${ROS_DISTRO}/setup.bash; catkin_make'

RUN mkdir -p /home/ifm/catkin_ws/ifm3d-ros/
RUN /bin/bash -c 'cd /home/ifm/catkin_ws/ifm3d-ros/; . /opt/ros/${ROS_DISTRO}/setup.bash; catkin_make install -DCMAKE_INSTALL_PREFIX=/home/ifm/catkin_ws/ifm3d-ros/install'
```

coa06yl, 2023-06-15:
@dekhanra It is my first time to use docker.  I managed to catkin_make it. In order to launch the camera, may I know the package name and launch file name? Thanks 
[image]

desengph, 2023-06-15:
Hi @coa06yl 
We use the Dockerfiles for two options:

+ To have a consistent user environment and dependency setup: i.e. check that the ROS node compiles and runs.
In your case, if you share the ROS master and node in the same Docker container you can treat it exactly the same as if ROS were running “bare metal” on your native machine - i.e regular ROS launch: https://github.com/ifm/ifm3d-ros/tree/v0.7.0#camera-nodelet
+ Alternatively, you can treat the Dockerfile as installation and setup “recipe” to configure your native machine equivalently and run it there.

coa06yl, 2023-06-19:
@dekhanra I managed to run the camera node. But from the docker, I couldn't ping the camera. Should I use port binding or some other methods to connect the camera? By the way, my ROS version is Noetic.
 
[image]
[image]

dekhanra, 2023-06-19:
Hi @coa06yl, can you able to ping the camera inside the docker container? To execute ping inside docker container you may need to install the `iputils-ping` package.
```bash
apt-get install -y iputils-ping
```
I guess you have given `--network` flag as `host` while running docker container

coa06yl, 2023-06-22:
@dekhanra after install iputils, I am not able to ping the camera inside docker. 
[image]

coa06yl, 2023-06-22:
@dekhanra when I try to bind the port, I encountered this msg. ros master port: 11311 camera port: 80
[image]

graugans, 2023-06-22:
Hej @coa06yl I guess the base error is still the same as: https://github.com/ifm/ifm3d-ros/issues/70 can you please verify with the ifm Vision Assistant if the camera is maybe put into software trigger mode?

coa06yl, 2023-06-22:
Hi @graugans ,  you mean this? Should be continuous?

[image]

dekhanra, 2023-06-22:
Hi @coa06yl, 

Is the `Timeout waiting for camera!` error continuous or just at startup of the ROS node? It is expected if it comes at startup due to latency and the node is not closed. You can increase the `timeout_millis` parameter in the `launch` file to 600 to 700 milliseconds which may help, but if it the error is continuous then we need to look further.

coa06yl, 2023-06-26:
@dekhanra I managed to connect to O3X camera and received data from it. I installed gedit in docker and tried to edit the camera.launch file in docker container. But didn't succeed. 
[image]

dekhanra, 2023-06-26:
@coa06yl,

It is expected during the Node startup because of Latency as I explained above. If the issue is resolved can we close this Issue
? If there are any further questions please feel free to open to new issue.

coa06yl, 2023-06-27:
Hi @dekhanra  Once again thank you for your help. Pls close this issue.

closed 2023-06-30 by coa06yl