ifm/ifm3d-ros #75: Startup time is too long, about 35s

FlyHan, 2023-07-25:
Startup time is too long, whether it can be optimized
One reboot took 10s. The logs:

> 07/25 08:28:54.334 [I][laser_hal][laser_launch.cpp:455] start cmd:roslaunch laser_hal ifm3d_o3x100.launch
07/25 08:28:58.906 [I][ifm3d/camera0][nodelet.cpp:196] Loading nodelet /ifm3d/camera0 of type ifm3d_ros/camera_nodelet to manager camera0_standalone_nodelet with the following remappings:
07/25 08:28:59.050 [I][ifm3d/camera0][service.cpp:80] waitForService: Service [/ifm3d/camera0_standalone_nodelet/load_nodelet] has not been advertised, waiting...
07/25 08:28:59.564 [I][ifm3d/camera0_standalone_nodelet][loader.cpp:233] Initializing nodelet with 4 worker threads.
07/25 08:28:59.680 [I][ifm3d/camera0][service.cpp:122] waitForService: Service [/ifm3d/camera0_standalone_nodelet/load_nodelet] is now available.
07/25 08:29:00.391 [I][ifm3d/camera0_standalone_nodelet][camera_nodelet.cpp:55] onInit(): /ifm3d/camera0
07/25 08:29:06.436 [I][ifm3d/camera0_standalone_nodelet][camera_nodelet.cpp:229] updateDiagnostic get in
07/25 08:29:06.437 [I][ifm3d/camera0_standalone_nodelet][camera_nodelet.cpp:686] ifm3d icmp check success
07/25 08:29:06.637 [I][ifm3d/camera0_standalone_nodelet][camera_nodelet.cpp:595] Running dtors...
07/25 08:29:14.438 [I][ifm3d/camera0_standalone_nodelet][camera_nodelet.cpp:602] Initializing camera...
07/25 08:29:15.584 [I][ifm3d/camera0_standalone_nodelet][camera_nodelet.cpp:607] Initializing swu_...
07/25 08:29:15.597 [I][ifm3d/camera0_standalone_nodelet][camera_nodelet.cpp:634] wait For Productive success...
07/25 08:29:15.597 [I][ifm3d/camera0_standalone_nodelet][camera_nodelet.cpp:638] Initializing framegrabber with mask 16...
07/25 08:29:17.110 [I][ifm3d/camera0_standalone_nodelet][camera_nodelet.cpp:640] Initializing image buffer...
07/25 08:29:17.612 [I][ifm3d/camera0_standalone_nodelet][camera_nodelet.cpp:567] Syncing camera clock to system...
07/25 08:29:18.436 [I][ifm3d/camera0_standalone_nodelet][camera_nodelet.cpp:210] o3x100 is soft on now, uptime is 88.000000, operating mode is Run
07/25 08:29:18.520 [I][ifm3d/camera0_standalone_nodelet][camera_nodelet.cpp:919] Got unit vectors, restarting framegrabber with mask: 15
07/25 08:29:18.521 [I][ifm3d/camera0_standalone_nodelet][camera_nodelet.cpp:595] Running dtors...
07/25 08:29:26.322 [I][ifm3d/camera0_standalone_nodelet][camera_nodelet.cpp:602] Initializing camera...
07/25 08:29:27.335 [I][ifm3d/camera0_standalone_nodelet][camera_nodelet.cpp:607] Initializing swu_...
07/25 08:29:27.348 [I][ifm3d/camera0_standalone_nodelet][camera_nodelet.cpp:634] wait For Productive success...
07/25 08:29:27.348 [I][ifm3d/camera0_standalone_nodelet][camera_nodelet.cpp:638] Initializing framegrabber with mask 15...
07/25 08:29:28.109 [I][ifm3d/camera0_standalone_nodelet][camera_nodelet.cpp:640] Initializing image buffer...

FlyHan, 2023-07-26:
Hi @lola-masson 

Thanks for the reply.
You are right. I'm using an O3X100, and talking about the start up of the ROS node.

firmware, ifm3d, ifm3d-ros, the versions are as follows:
>  firmware: IFM_Software = 1.1.190",
    ifm3d:  version=0.18.0 
    ifm3d-ros: 0.6.2

lola-masson, 2023-07-26:
I just tried it on my setup, using the same firmware as yours, with ifm3d 0.20.3 and ifm3d-ros 0.7.0. I see that the time between the roslaunch command and the data actually streaming is 7 seconds. Could you explain how you get the 35 seconds you mention?

FlyHan, 2023-07-27:
Hi @lola-masson 

Thanks for trying.
The 35s I mentioned were analyzed through the ROS logs mentioned on the first floor, from **08:28:54.334** to **08:29:28.109**.
Can you share ROS logs for 7s?  I'll try ifm3d-ros 0.7.0 myself to compare it

FlyHan, 2023-07-27:
Hi @lola-masson

I found the problem, the InitStructures() function is delayed by 8s, so that the power-on startup delay is 16s.
The delay of 8s is added because the abnormal automatic restart function is added, and the delay is required after the restart to connect successfully.

Thank you so much

closed 2023-07-28 by FlyHan