ifm/ifm3d-ros #76: RPC Response Indicates Failure: XMLRPC Method Not Found (D-Bus Error 16)

Epik-Robotik, 2024-06-21:
Additional Information
Environment:
ifm3d version: v1.5.3
ifm3d-ros version: v1.1.2
camera: O3DP03
ros-noetic, ubuntu:20.04

**### Description**

I encountered an issue while trying to run the ifm3d_ros_driver nodelet. The error log indicates an RPC response failure due to the XMLRPC method not being found, accompanied by a D-Bus error 16. Below are the detailed error messages and logs:

### **Error Logs**
```
2024-06-21 09:39:16 ERROR [/home/user/ifm3d_ws/src/ifm3d-1.5.3/modules/device/src/libifm3d_device/xmlrpc_wrapper.hpp:100] http://192.168.0.69/api/rpc/v1/com.ifm.efector/ -> get: RPC response indicates failure. XMLRPC method not found (D-Bus error 16)
2024-06-21 09:39:16 WARN [/home/goktug/ifm3d_ws/src/ifm3d-1.5.3/modules/device/src/libifm3d_device/o3r_impl.hpp:221] XMLRPC ERRO: Unknown error 100001: XMLRPC method not found (D-Bus error 16)
2024-06-21 09:39:16 ERROR [/home/user/ifm3d_ws/src/ifm3d-1.5.3/modules/device/src/libifm3d_device/xmlrpc_wrapper.hpp:100] http://192.168.0.69/api/rpc/v1/com.ifm.efector/ -> get: RPC response indicates failure. XMLRPC method not found (D-Bus error 16)
[FATAL] [1718951956.842165960]: Failed to load nodelet '/ifm3d_ros_driver/camera of type ifm3d_ros/camera_nodelet to manager camera_standalone_nodelet'
[ifm3d_ros_driver/camera-3] process has died [pid 114274, exit code 255, cmd /opt/ros/noetic/lib/nodelet/nodelet load ifm3d_ros/camera_nodelet camera_standalone_nodelet name:=camera log:=/home/goktug/.ros/log/fa9e3a82-2f98-11ef-b10f-99868ddf861f/ifm3d_ros_driver-camera-3.log].
log file: /home/user/.ros/log/fa9e3a82-2f98-11ef-b10f-99868ddf861f/ifm3d_ros_driver-camera-3*.log

```

**### Steps to Reproduce**
Setup the ifm3d environment using the provided setup scripts.
Run the ifm3d_ros_driver nodelet using the following command:

        `roslaunch ifm3d_ros_driver camer_3d.launch`
Observe the terminal output and logs for the mentioned error.
**### Expected Behavior**
The ifm3d_ros_driver nodelet should load successfully without any XMLRPC or D-Bus errors.

**### Actual Behavior**
The nodelet fails to load with RPC response indicating failure due to the XMLRPC method not being found and a D-Bus error 16.

ROS Distribution: Noetic
ifm3d Version: 1.5.3
OS: Ubuntu 20.04
Network Configuration: Static IP 192.168.0.69 for the device
Related Log Files: ifm3d_ros_driver-camera-3.log
Request for Help
I would appreciate any guidance on resolving this issue. If there are any additional diagnostics or logs needed, please let me know.

dekhanra, 2024-06-21:
Hello @Epik-Robotik 

The current tagged version you are using is only compatible with O3R cameras but not for  O3DPxx cameras,

Epik-Robotik, 2024-06-21:
Which version is suitable for O3DPxx cameras? Can you make a suggestion?

Epik-Robotik, 2024-06-21:
@dekhanra Which version is suitable for O3DPxx cameras? Can you make a suggestion?

dekhanra, 2024-06-21:
Hello @Epik-Robotik, we don't have any ROS wrappers compatible with O3DPxx cameras. Please use pds-cppclient to implement the ROS wrappers.